DENSORobot/denso_robot_ros #2: Support for VS6577 and IO control

Booooooosh, 2017-12-08:
Thanks very much for making efforts to provide a ROS interface with DENSO robot!

I know that this package is under development. Just curious whether this package serves as a generic controller for all DENSO robots with 6DOF? Actually I'm using VS6577 instead of VS060, will this package still work?

And I'll need I/O control in order to start up my end effector. Will this package provide this interface to the robot?

Many thanks!

YoshihiroMIYAKOSHI, 2017-12-12:
Sorry for my late reply.

Our denso_robot_ros package supports VS6577, but there is only a VS060 URDF as a standard component.
As I said in the start-jsk's issue (https://github.com/start-jsk/denso/issues/89), to create other robot types URDF, you can use ROSConverter tool.
Refer to the http://wiki.ros.org/denso_robot_ros/ROSConverter page for more details.

Our package does not have the I/O control function just now. I'll add the function in this week.

Booooooosh, 2017-12-12:
Many thanks for the reply!
Yes, I've got my own model for VS6577 and just waiting for the I/O functionality!

Many thanks!

closed 2017-12-12 by Booooooosh