DENSORobot/denso_robot_ros #3: Run DENSORobot RC8-VP6242 and ROS

lacduong, 2017-12-23:
Dear @YoshihiroMIYAKOSHI @gavanderhoorn 

I would like to run a real Denso robot from your denso_robot_ros packages.

I have been imported the VP6242_description to the ros_package. It ok when simulating with gazebo.

But when I run with a real robot, It cant run and issue many issues.

[image]

This is the list of Errors:
(1) ERROR] [1514014457.959615445]: Invalid robot type

[image]

(2) [ERROR] [1514014475.056066194]: Action client not connected: vp6242a/arm_controller/follow_joint_trajectory

[image]

(3) WARNING

[image]

(4) [ERROR] [1514014834.840473065]: Unable to identify any set of controllers that can actuate the specified joints: [ joint_1 joint_2 joint_3 joint_4 joint_5 joint_6 ]
[ERROR] [1514014834.840529921]: Known controllers and their joints:

[image]

I would like to receive your help to fix the errors to run with a real robot.

Thank you very much.
Best regards,
Lac Duong

lacduong, 2017-12-23:
I think the main problem that is:
**Controller Spawner: Waiting for service controller_manager/load_controller**
**[WARN] [1514042089.481948]: Controller Spawner couldn't find the expected controller_manager ROS interface**

[image]

Please help me, your assistance is appreciated!

lacduong, 2017-12-24:
**Update: I fixed all above issues, But I occur a problem:** Please help me
[ INFO] [1514108526.644151677]: initialize bCap slave connection
[ INFO] [1514108526.644261311]: try to fill buffer
[ WARN] [1514108526.664453525]: failed to slvmove, errorcode: 8150001b
[FATAL] [1514108526.664509621]: failed to fill bcap buffer
[ERROR] [1514108526.664540755]: failed to initialize denso robot

YoshihiroMIYAKOSHI, 2017-12-24:
I checked this issue. Our denso_robot_control package checks the robot type.
Your launch file expects the robot name is "VP-6242 A", but the robot controller only returns "VP-6242" even if that has the safety option.
I fixed the checking robot name function in denso_robot_control, but It will  take times to update ROS repositories one. Could you edit your launch file to be enabled connect real robot?
before: param name="robot_name" value="VP-6242 A"
after: param name="robot_name" value=**"VP-6242"**

lacduong, 2017-12-25:
Thank @YoshihiroMIYAKOSHI  very much. I already run the real robot with your package!

YoshihiroMIYAKOSHI, 2017-12-25:
If this issue already solved, could you close this issue?

closed 2017-12-25 by lacduong

mikami235, 2018-01-12:
Dear @lacduong I have the similar problem with " [ERROR] [1514014475.056066194]: Action client not connected: vp6242a/arm_controller/follow_joint_trajectoryyour Error(2)"
How did you solve the problem?
I tried to use denso package for denso real robot (different from VS060 and VP-6242)

Best regards

lacduong, 2018-01-13:
@mikami235 

Do you have the error Invalid Robot type? -> if this error: the cause may be the wrong of robot name

Best
Duong Van Lac

mikami235, 2018-01-13:
@lacduong 
I appreciate your quick reply.
Yes. I have the error "Invalid robot type" , too, after I renamed both the file name and <arg name="robot_name"> regarding "vs060_bringup.launch" and files in "vs060_description".
And I prepared the configuration file "***.urdf"

However, "invalid robot type" has remain

Should I rename all files including "vs060"?

lacduong, 2018-01-13:
@mikami235 

Do you use the real robot vs60 or vp6242?
You only have to change the robot name in the urdf file. 
param name="robot_name" value="VP-6242"

Be careful, the real robot and the robot_description need the same.
Best,

mikami235, 2018-01-14:
@lacduong
I appreciate your advise.

Best regarg.

jshuai, 2018-01-23:
Dear YoshihiroMIYAKOSHI
Can you share me your URDF about VP6242？
I will be very grateful to you。
Thanks。

lacduong, 2018-01-24:
@jshuai 
For your request email, I would like to send you the VP6242A description.
[vp6242a_description.zip](https://github.com/DENSORobot/denso_robot_ros/files/1658252/vp6242a_description.zip)

mikami235, 2018-01-28:
@lacduong
I was able to launch "VS-068" by rewriting like the below;
<param name="robot_name" value="VS068A4-AV6-NNN-NAN A"/>

Be careful to input characters after "VS068" because they are depending on controller specification.

Thank you very much

Best regard