DENSORobot/denso_robot_ros #4: Not able to spawn the controller when using new model (using sim:=true)

Booooooosh, 2018-01-20:
My headache
---
I built my own model for VS6577 and when I tried to bring up the whole system in simulation mode, it says:

**1. Failed to call service get_planning_scene, have you launched move_group? at /tmp/binarydeb/ros-indigo-moveit-ros-planning-0.7.13/planning_scene_monitor/src/planning_scene_monitor.cpp:480**
**2. [ERROR] [1516414069.030088645, 15.171000000]: MoveItSimpleControllerManager: Action client not connected: vs6577/arm_controller/follow_joint_trajectory**

So here is what I did to replace the VS060 with the VS6577:
---
1. I created my own vs6577_description under denso_robot_descriptions package, where I have stl files for VS6577 and urdf file and **denso_robot_control.yaml and vs6577.launch.xml (Here is a question: what are these two files? I assume the first yaml file is for providing params for spawning controllers and I'm really confused about the launch file here, should I change something?)** 

2. I created my own vs6577_config under denso_robot_moveit_config package, where I have controllers.yaml, joint_limits.yaml and vs6577.srdf (the latter two are generated my moveit_setup_assistant)

3. I have my own vs6577_bringup.launch which is basically the same as vs060_bringup.launch except that I changed the robot_name from vs060 to vs6577.

Any suggestion?
---
I can provide extra information if you need it. I feel like this is just because I'm not understanding some of the parameters in some yaml file.

**Many thanks again for providing this interface and time for looking at my issue!!!**

YoshihiroMIYAKOSHI, 2018-01-20:
Why don't you use our ROSConverter (http://wiki.ros.org/denso_robot_ros/ROSConverter)?
We recommend to use ROSConverter for creating new robot type rather than create by own.

Booooooosh, 2018-01-20:
Many thanks! I will try that and come back to you.

Booooooosh, 2018-01-27:
This might be a stupid question, but when I downloaded the WINCAPS III installer, do I need a CD to install the software? Because right now when I click "install", it says that "Can Not Find Installer".

YoshihiroMIYAKOSHI, 2018-01-28:
It does not need CD to install. I think the message may be shown when you failed to download zip file. Could you re-download WINCAPS III?
or you do not need install the latest WINCAPS III to use ROSConverter. The ROSConverter is located in the unzipped folder. So you can use it by only download the latest WINCAPS III.

Booooooosh, 2018-01-28:
I solved it, I installed another version of WINCAPS and upgraded to v3.55.0 successfully. I will try the model and let you know the result.
Thanks anyway!!

Booooooosh, 2018-01-28:
Hi, I think it is working. Many thanks!
However, I came across another issue. I guess I will close this issue and start another one to be clear.
Many thanks!

closed 2018-01-28 by Booooooosh