DENSORobot/denso_robot_ros #5: Controller ABORTED

Booooooosh, 2018-01-28:
Every time I execute the plan. I got the following warning:

**[ WARN] [1517105508.508283943, 10.232000000]: Dropping first 1 trajectory point(s) out of 21, as they occur before the current time.
First valid point will be reached in 0.604s.
[ WARN] [1517105513.940435591, 14.934000000]: Controller vs6577/arm_controller failed with error code GOAL_TOLERANCE_VIOLATED
[ WARN] [1517105513.940549747, 14.934000000]: Controller handle vs6577/arm_controller reports status ABORTED**

- I think the first warning is fine, it has something to do with the implementation of MoveIt itself.
- While the second and the third one is my concern: the controller seems to suffer a GOAL_TOLERANCE_VIOLATED every time it carries out a plan. It turns out that the plan is executed (the manipulator want to the desired position successfully), however, the controller still reports ABORTED and the return value by the execute() function is FALSE.
- I'm not quite familiar with ros_controller, have you ever met this problem?
- I'm using Ubuntu14.04 with ROS INDIGO

Many thanks!!

YoshihiroMIYAKOSHI, 2018-02-01:
Sorry for my late reply. Unfortunately I have not met this problem.
Did you launch vs6577_bringup.launch with "sim:=false" option when you met this problem?

Booooooosh, 2018-02-02:
Yes. I guess it has something to do with my Gazebo. Since when I interface with the real robot, it's working fine. Are you using Gazebo 7?
Many thanks for the reply!

YoshihiroMIYAKOSHI, 2018-02-07:
I had checked denso_robot_bringup with Gazebo 2 (ROS default).
Are there any other warnings/errors when you launch?

Booooooosh, 2018-02-07:
Nope, just this one. It's working fine with the real robot, though.

Booooooosh, 2018-02-07:
Oh, I have another question (it's not an issue, so I guess I just ask here): how do I change the max velocity when connecting to the real robot? Should I change the value within the joint_limits.yaml (in there is a parameter called "max velocity").
Many thanks!

YoshihiroMIYAKOSHI, 2018-02-10:
Yes, I think you can change the max velocity by editing the joint_limit.yaml.
has_velocity_limits to true and max_velocity to the velocity which you want.
Does there have no effect?

Booooooosh, 2018-02-10:
Nope. I tried it yesterday, is it because I'm using cartesian_plan?

YoshihiroMIYAKOSHI, 2018-02-11:
Do you use computeTimeStamps?
I found an issue about cartesian plan. (https://groups.google.com/forum/#!topic/moveit-users/x5FwalM5ruk)
Sorry, I don't know MoveIt! deeply, so my advice may be wrong.

Booooooosh, 2018-04-18:
OK, finally I figured this out. Turns out that changing joint_limit.yaml does do the trick, the previous problems lies in the git version control.

closed 2018-04-18 by Booooooosh

desstiony, 2020-06-23:
> 好，终于我明白了。事实证明，更改joint_limit.yaml确实可以解决问题，以前的问题在于git版本控件。

The same problem I encountered, how did you solve it

felix-wt, 2020-07-01:
> OK, finally I figured this out. Turns out that changing joint_limit.yaml does do the trick, the previous problems lies in the git version control.

hi,  I have the same problem.Could u tell me how to solve it.

felix-wt, 2020-07-01:
> OK, finally I figured this out. Turns out that changing joint_limit.yaml does do the trick, the previous problems lies in the git version control.

@Booooooosh 
hi, I have the same problem.Could u tell me how to solve it.

LnBdd, 2021-03-03:
@Booooooosh @YoshihiroMIYAKOSHI 
Coming back to the original issue:
I just installed the melodic version of denso_robot_ros on Ubuntu 18.04 with ROS melodic and run into the same issue (vs060/arm_controller failed with error GOAL_TOLERANCE_VIOLATED) when starting the vs060 model provided by the package ( 'vs060_bringup.launch sim:=true').
The robot moves to the desired end position but due to the controller error, it does not completely stop but keeps moving very slowly. 
I run into this issue without any changes on the original files. However, testing multiple possible fixes provided by similar issues (such as: increasing the goal tolerance or increasing the allowed_execution_duration_scaling) did not lead to a solution. 
Any solutions? Am I missing something?

furukawahiromi, 2021-03-05:
I pushed feature/#5.
https://github.com/DENSORobot/denso_robot_ros/tree/feature/%235

ref.
GOAL_TOLERANCE_VIOLATED (Gazebo/Moveit) https://answers.ros.org/question/312123/goal_tolerance_violated-gazebomoveit/

DensoWaveRobot, 2021-03-08:
>

> I pushed feature/#5.
> https://github.com/DENSORobot/denso_robot_ros/tree/feature/%235
> 
> ref.
> GOAL_TOLERANCE_VIOLATED (Gazebo/Moveit) https://answers.ros.org/question/312123/goal_tolerance_violated-gazebomoveit/

In this commit, stopped_velocity_tolerance is added to the parameter as 0.

commit 8b74a3db (2021-03-10): Fixed #5 - Controller ABORTED

  - Add stopped_velocity_tolerance to denso_robot_control.yaml.