DENSORobot/denso_robot_ros #6: Failed to write. (83500121)

Booooooosh, 2018-04-18:
Sometimes when the manipulator is executing the trajectory, this error occurs and I get disconnected automatically, any idea what's the possible cause for this?

ToshitakaSuzuki, 2018-04-20:
This error occurs when a slave command is issued even though the slave mode is off.
Unfortunately, I can't identify what caused the slave mode turned off from your comment.
What kind of error message was displayed on the teach pendant?

ToshitakaSuzuki, 2019-02-28:
@Booooooosh 
Are there any progress with this?

Booooooosh, 2019-02-28:
Sorry for the late reply. Unfortunately I have graduated and no longer in charge of this project. IIRC, the teaching pendent is complaining about slave command is sent while slave mode is off.
This issue kinds of happens randomly, so I can not pinpoint the issue easily, sorry...

Booooooosh, 2019-02-28:
I will close this issue for now.

closed 2019-02-28 by Booooooosh

VideoSystemsTech, 2020-11-04:
Hi,
I have the same error on my VS-087.
It also happens to me randomly when executing the trajectory.
I can run some tests if it is needed.

DensoWaveRobot, 2020-11-04:
Is there any other error in RC8 at the same time or right before? 
RC8 change slave mode to normal mode, when some error occurs.
You can check error history on teaching pendant [F6 setting]-[F4 operation and error log].

VideoSystemsTech, 2020-11-04:
The error is this:

```
Position command buffer is empty  
[ OriginalNumber] : 84201482

[ Description ] : Position command was not generated for a certain period of time.
[ Recovery ] : If the system is set to Slave move, check that the position command is correctly sent from the master side.
```

VideoSystemsTech, 2020-11-05:
Sometimes the error of issue #12 is raised, so I think these two issues are related.
This seems to happen when the `max_velocity_scaling_factor` and/or `max_acceleration_scaling_factor` are low.
The error is raised when both are set to `0.1` for example, but it is not raised when set to `0.15`.

DensoWaveRobot, 2020-11-06:
@VideoSystemsTech 
Thank you for your reply.
Denso robot is change mode to slave mode when it receive move operation from moveit node.
Robot in slave mode processes these operation Periodically. 
During this time, If there are no operation in robot buffer, the error 84201482 occurs.
it often happen caused by communication delay between ROS PC and robot.
Could you try to fix code as follows?

denso_robot_ros/denso_robot_control/src/denso_robot_control.cpp
Line 57  rate.sleep(); ←comment out

Please Re-setup denso_robot_ros after fix it.

DensoWaveRobot, 2020-11-06:
The error of issue #12 80000900 is generic error for denso_robot_ros.
So I think root cause is 84201482.

VideoSystemsTech, 2020-11-09:
> @VideoSystemsTech
> Thank you for your reply.
> Denso robot is change mode to slave mode when it receive move operation from moveit node.
> Robot in slave mode processes these operation Periodically. [...]

That doesn't change anything.

DensoWaveRobot, 2020-11-12:
@VideoSystemsTech 
Thank you for you trying and sorry I couldn't help you.

I have one more thing to check.
When scale_factor cahnged, it often happen that relate with velocity or acceleration.
There are no errors about these?
If there are not, it due to neteork traffic.
Traffic is often caused by deficiency in performance of PC, repeater or extender.
(CPU load by RViz monitor and so on)

If traffic cause the error, you may be able to solve by changing cobotta slavemode synchronous to asynchronous.

About difference of slavemode, please refer to this document in folder ORiN2 or WINCAPS intalled.
"C:\ORiN2\CAP\b-CAP\CapLib\DENSO\RC8\Doc\b-CAP_Guide_RC8_en.pdf"

Overview of how to change slavemode

1.Do the follows 
ubuntu@roslaunch denso_robot_bringup <launch file of ROBOT> sim:=false ip_address:=<ROBOT's IP Address>
2.Do the follows on new terminal
ubuntu@roslaunch bcap_service bcap_service.launch ip_address:=<ROBOT's IP Address>
3.Do the follows on new terminal
ubuntu@rosservice call /bcap_service '{func_id: 3, vntArgs: [{vt: 8, value: "b-CAP"}, {vt: 8, value: "CaoProv.DENSO.VRC"}, {vt: 8, value: "localhost"}, {vt: 8, value: ""}] }'
4."<controller handle>" is returned as follows
HRESULT: 0 vntRet: vt: 19 value: "<controller handle>"
5.Do the follows using "<controller handle>"
ubuntu@rosservice call /bcap_service '{func_id: 7, vntArgs: [{vt: 3, value: "<controller handle>"}, {vt: 8, value: "Robot0"}, {vt: 8, value: ""}] }'
6."<robot handle>" is returned as follows
HRESULT: 0 vntRet: vt: 19 value: "<robot handle>"
7.Do the follows using "<robot handle>"
ubuntu@rosservice call /bcap_service '{func_id: 64, vntArgs: [{vt: 3, value: "<robot handle>"}, {vt: 8, value: "slvChangeMode"}, {vt: 8, value: "0x102"}] }'
func_id:64 // ID_ROBOT_EXECUTE
vt:3 // VT_I4 (LONG)
vt:8 // VT_BSTR

*attention
During slavemode is asynchronous, no error occurs but operation delay by traffic may make robot movement jerky.
In this case, you have to slow down velocity.

DensoWaveRobot, 2020-11-13:
@VideoSystemsTech 
Sorry. You can change slave mode in easier way.
Please use below method instead of above.

1.Do the follows
$ rostopic pub /<robot name>/ChangeModetd_msgs/Int32 "data: 0"
2.Press Ctrl+C button to stop command
3.Do the follows
$ rostopic pub /<robot name>/ChangeModetd_msgs/Int32 "data: 0x102"
4.Press Ctrl+C button to stop command

k-okada, 2020-12-17:
@DensoWaveRobot , May be you missed Copy & Paste
```
1.Do the follows
$ rostopic pub //ChangeMode std_msgs/Int32 "data: 0"
2.Press Ctrl+C button to stop command
3.Do the follows
$ rostopic pub //ChangeMode std_msgs/Int32 "data: 0x102"
4.Press Ctrl+C button to stop command
```

DensoWaveRobot, 2020-12-17:
@k-okada 
Sorry, I made some typo.
Thank you for you comment.

suc1, 2021-09-08:
I am using denso SCARA robot arm (RC8 HSR048A1). The ubuntu bionic (18.05.5) is running on Virtual Box 6.1.
Following the tutorial in
[denso_robot_ros Tutorials](http://wiki.ros.org/denso_robot_ros/Tutorials)
It seeems **SlvChangeMode()** cause the problems. The bug always happens.

when run
`roslaunch denso_robot_bringup line8left_bringup.launch sim:=false ip_address:=192.168.0.1`
Get errors:
>[ INFO] [1631121744.315385728]: bcap-slave timeout changed to 16 msec [mode: 0x202]
>[ERROR] [1631121744.564034131]: Failed to write (83500121)
>[ERROR] [1631121744.567636151]: Automatically change to normal mode.
>...
>[ERROR] [1631121745.020626639]:   [2] You cannot execute a command in Slave mode (83501032)
>[ERROR] [1631121745.024187088]:   [1] SlvMove command was executed while the Slave mode was not selected. (83500121)

In TP, the 1st error is:
>You cannot execute a command in slave mode
>[OriginalNumber] 83501032
>[Description] This operation is not available while the robot is in Slave mode
>[Recovery] This command is not available under the Slave mode(range bounded by SlvChangeMode command). Run the command at the outside of the area surround by SlvChangeMode

The 2nd error is:
>SlvMove command was executed while the slave mode was not selected
>[OriginalNumber] 83500121
>[Description] SlvMove command was executed when slave mode was not selected
>[Recovery] Switch to the Slave mode and retry

Because my robot is not the default in this tutorial, line8left was converted by ROSConverter/AcquireVelAcc