DENSORobot/denso_robot_ros #8: Failed to connect real controller. (80000900)

flyinskybtx, 2019-01-04:
Error when running 
`roslaunch denso_robot_bringup denso_robot_bringup.launch sim:=false ip_address=<vs060 address>`
The problem is 

> Failed to connect real controller. (80000900)

the error is raised in running
`  <node name="denso_robot_control" pkg="denso_robot_control" type="denso_robot_control" output="screen" ns="/$(arg robot_name)"/>`

that in denso_robot_hardware.cpp that
`    HRESULT hr = m_eng->Initialize();
if(FAILED(hr)) {
  ROS_ERROR("Failed to connect real controller. (%X)", hr);
  return hr;`
throws the exception. However the error code varies if I launch the project seperately or in a roll. I have experienced (80000900) or (80000909) or something else like 8000XXXX.

I am trying to control my real VS060 robot with RC8A controller through ROS, which is kinetic version on  ubuntu16.04 and i have already registered the license for b-cap as suggested in the tutorial http://wiki.ros.org/denso_robot_ros/Tutorials/How%20to%20control%20an%20RC8%20with%20MoveIt%21
and converted the vs060-descriptions. I have tried both the apt-get version or source built  kinetic-devel version of this repo, but facing the same Failed to connect real controller problem.

could you provide some thoughts about how could this problem happen? 
thanks in advance and wish u a happy new year!

gavanderhoorn, 2019-01-04:
Keeping things connected: [denso ros failed to connect real controller](https://answers.ros.org/question/311912) (ROS Answers).

flyinskybtx, 2019-01-04:
> Keeping things connected: [denso ros failed to connect real controller](https://answers.ros.org/question/311912) (ROS Answers).
Thanks for your information!

I am not clear what do you mean by keep connected. The fact is that it is able to ping the robot from my computer and vice versa, but I dont know if my computer is able to connect the robot through b-cap services. Do you have any idea how can I check the connection?

and i am also confused where to ask the questions. i do posted the same questions in both the issue and ros.answers. and should i close one of them?

gavanderhoorn, 2019-01-04:
> I am not clear what do you mean by keep connected. 

You posted your issue to both this tracker as well as ROS Answers. That is called cross-posting. In order to avoid information getting lost (because it is either only posted on ROS Answers or this tracker), I commented with a link to your ROS Answers question so future readers would be able to find the post there as well.

> and i am also confused where to ask the questions. i do posted the same questions in both the issue and ros.answers.

That would be something for the maintainer(s) to answer. I don't know what they prefer.

flyinskybtx, 2019-01-04:
> You posted your issue to both this tracker as well as ROS Answers. That is called cross-posting. In order to avoid information getting lost (because it is either only posted on ROS Answers or this [...]

thanks for you reply, 
i am going to close this issue and post my updates in ros.answers

and i updated my questions that locates the error code should be 80000909, which means "packet is too much", do you have any idea how this happens?

closed 2019-01-04 by flyinskybtx