DENSORobot/denso_robot_ros #9: Fail to connect real controller (80000909)

flyinskybtx, 2019-01-07:
I've been working around this problem for three days, and continue trying to figure out the solution. After struggled into the docs and code of  b-Cap and ORiN,  i decided to post the problem here, instead of answers.ros.org, because I think this problem is more a b-Cap connection problem than ROS problem. 

The error occurs when I run the tutorial according to:
http://wiki.ros.org/denso_robot_ros/Tutorials/How%20to%20control%20an%20RC8%20with%20MoveIt%21
and the real controller cannot be connected, and raise the following issue:
`[ERROR] [1546830111.567536479]: Failed to connect real controller. (80000909)`

the ros params are using the project default, as:
` * /robot_description: <robot name="vs06...
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /vs060/arm_controller/joints: ['joint_1', 'join...
 * /vs060/arm_controller/type: position_controll...
 * /vs060/arm_group: 0
 * /vs060/config_file: /home/flyinsky/Re...
 * /vs060/conn_type: tcp
 * /vs060/controller_name: 
 * /vs060/controller_type: 8
 * /vs060/invoke_timeout: 180000
 * /vs060/ip_address: 192.168.1.33
 * /vs060/joint_1: 1
 * /vs060/joint_2: 1
 * /vs060/joint_3: 1
 * /vs060/joint_4: 1
 * /vs060/joint_5: 1
 * /vs060/joint_6: 1
 * /vs060/joint_state_controller/publish_rate: 125
 * /vs060/joint_state_controller/type: joint_state_contr...
 * /vs060/port_number: 5007
 * /vs060/recv_format: 292
 * /vs060/retry_count: 5
 * /vs060/robot_joints: 6
 * /vs060/robot_name: VS060A3-AV6-NNN-NNN
 * /vs060/send_format: 288
 * /vs060/timeout: 3000
 * /vs060/wait_time: 1
 * /vs060/watchdog_timer: 400
`
I have checked the error code, which should be , according to http://www.orin.jp/e/wp-content/uploads/sites/2/2016/05/ORiN2_ProgrammersGuide_en.pdf and https://github.com/fsuarez6/bcap/blob/master/include/bcap/dn_common.h, the problem should be the connecting **packet on UDP is too big.** 
another problem could be that my ROS-kinetic is built on python2.7 as recommended. but when I try the python-b-cap version according to https://github.com/ShoheiKobata/orin_bcap_python_samples , only if i use my python 3 interpreter can be able to connect controller and successfully run the SlaveMode. Python 2.7 throw exception that probably String type issue.

In addition, when I test b-cap using b-cap test, the UDP connection throws the same 80000909 packet size error, but when I switch to TCP, the connection process is fine. However, when I modify the launch file bcap_service.launch.xml to set to TCP, and rosparams are changed, the connection still throws the 80000909 error. I dont know if the launch really uses the TCP connection.

And here is my question, what could possibly cause this error? How can I define the UDP packet size? could python2.7 be compatible with the b-cap service?

YoshihiroMIYAKOSHI, 2019-01-11:
Sorry for my late reply.
I found a bug, which cause the 80000909 error, in denso_robot_core.
The 80000909 error will happen when there are many programs in RC8.

I will fix this bug soon, but I think it will take times to update ROS repository's one.

flyinskybtx, 2019-01-17:
> Sorry for my late reply.
> I found a bug, which cause the 80000909 error, in denso_robot_core.
> The 80000909 error will happen when there are many programs in RC8.
> 
> I will fix this bug soon, but I think it will take times to update ROS repository's one.

Thanks for your reply!

Yes, after I have cleared all my RC8 programs (which not tool many though), I have been able to control the vs060 in b-caps mode.

looking forward to your updates

great jobs by the way, really helped me out! :)

ToshitakaSuzuki, 2019-03-04:
@flyinskybtx 
We fixed this bug and updated repositories.

allenwoods, 2019-04-12:
@YoshihiroMIYAKOSHI 
May I ask which PR or commit fixed this bug?

ToshitakaSuzuki, 2019-04-16:
@allenwoods 
This one. #11

allenwoods, 2019-04-18:
I see. Thank you 😁

closed 2021-05-12 by ToshitakaSuzuki