DENSORobot/denso_robot_ros #12: Failed to write (80000900)

flyinskybtx, 2019-02-21:
When i control the VS060  through bcap Slave Mode, sometimes i face the "failed to write (80000900)" error the mode then switched to Normal Mode.
The error occurs not very often, and unexpectable, It could happen at any step.
I think maybe its an connection problem, however i even tried to connect the computer directly with RC8A, and the same problem sometimes happens.

any idea what can i do about it?
THANKS!

zhoubo2018, 2019-02-22:
Hello, I also has this problem with every time to start up the cobotta robot with RC8 controller, I checked code, there are problems in "HRESULT hr" initial problem, why it happened?could you give me some suggestions?thanks.

ToshitakaSuzuki, 2019-02-22:
Can you check the error log from [Setting] -> [Operation and Error Log] in your teach pendant?
Are there any other errors besides [80000900] at the same time?

zhoubo2018, 2019-02-22:
> Can you check the error log from [Setting] -> [Operation and Error Log] in your teach pendant?
> Are there any other errors besides [80000900] at the same time?

my language is Chinese, the ERROR LOG it says like "not the output the area"

zhoubo2018, 2019-02-22:
> Can you check the error log from [Setting] -> [Operation and Error Log] in your teach pendant?
> Are there any other errors besides [80000900] at the same time?

the introduction for this LOG is I do some output to I/O that isn't in output area, but I do nothing to the robot, I use a standard configuration cobotta, and just run the package, nothing else, could you give me some suggestions?thank you very much

ToshitakaSuzuki, 2019-02-22:
@zhoubo2018 

> the introduction for this LOG is I do some output to I/O that isn't in output area, but I do nothing to the robot, I use a standard configuration cobotta, and just run the package, nothing else, could you give me some suggestions?thank you very much

Can you open the bringup.launch file of your COBOTTA and change the default values of following parameters?

- The default value of "send_format": 0

- The default value of "recv_format": 2

zhoubo2018, 2019-02-22:
> @zhoubo2018
>
>
> Can you open the bringup.launch file of your COBOTTA and change the default values of following parameters?
>
>     * The default value of "send_format": 0
>
>     * The default value of [...]

thank you for updating, I have changed that parameters to "0" and "2",
then the ros error is also the same, but in ERROR LOG is different, it says"first joint:position is out of software limit", the information about LOG says"because of tempt to excute the position out of software limit, all control stops, motor shutdown.please operate the robot move to the other direction".how should I continue?

ToshitakaSuzuki, 2019-02-22:
@zhoubo2018
Thank you for your reply.
>it says"first joint:position is out of software limit"

Did you launch the denso_robot_bringup after doing AutoCAL?
When you do AutoCAL, COBOTTA moves to almost software limit position.
It might cause "out of software limit" error.
Can you move COBOTTA to somewhat inside software limit by teach pendant, and then launch the denso_robot_bringup?

flyinskybtx, 2019-02-22:
> Can you check the error log from [Setting] -> [Operation and Error Log] in your teach pendant?
> Are there any other errors besides [80000900] at the same time?

Hi，
my error code in the TP is 84201482, which reads "delay in generating the commands" (it is writen in Chinese) and my ROS throws out the ' failed to write(80000900)' error. 
how can i fix this? 
thanks for reply

flyinskybtx, 2019-02-22:
> thank you for updating, I have changed that parameters to "0" and "2",
> then the ros error is also the same, but in ERROR LOG is different, it says"first joint:position is out of software limit", the [...]

First of all, you need to be aware that the ROS uses rad as rototional degree, while DENSO use degree angle. which means in ROS the angle should be 1.57, 3.14 while in DENSO those are 180, 360. 
If you got this right, I think you need to lookup into your moveit config to discover if the joint limits are right. and check the setting in your TP.

PS:  I am also Chinese, so you are welcome to contact me in Chinese, and if convenient, via WeChat: account: 757121348 name: flyinsky。 
i am currently working on  a DENSO vs060 robot arm.

zhoubo2018, 2019-02-25:
now I bring up the arm successfully, but when I use moveit to do some plan, the same error as flyinskybtx would occur, I use I met "the delaying to generate a command(84201482)"the information says "during a time , the system not  generate the command", how will be that? the moveit cannot use?

ToshitakaSuzuki, 2019-02-28:
@flyinskybtx 

> my error code in the TP is 84201482, which reads "delay in generating the commands" (it is writen in Chinese) and my ROS throws out the ' failed to write(80000900)' error.
how can i fix this?

Sorry for the late reply.
I cannot find the exact cause of your error.
I guess the communication delay occurs in some modules ( between COBOTTA and your PC, ubuntu, or ROS modules).

flyinskybtx, 2019-02-28:
> @flyinskybtx
>
>
> Sorry for the late reply.
> I cannot find the exact cause of your error.
> I guess the communication delay occurs in some modules ( between COBOTTA and your PC, ubuntu, or ROS modules).

Thanks for your suggestions!

I have the same doubt with you, if I connect the robot directly with a cable with my ubuntu, I can find a decrease in the times of that problem. However still remains. 
Now I work around with it by catching the exception and motor on again. (apparently not a good way to do it)
I found that other people re-write the ROS communication protocal, where they found that standard ROS communication could cause some problem with industrial robots. could that be the case?

ToshitakaSuzuki, 2019-03-01:
@flyinskybtx 
> I found that other people re-write the ROS communication protocal, where they found that standard ROS communication could cause some problem with industrial robots. could that be the case?

Thank you! I didn't know that.
Could you give us more information about this re-writing protocol matters (any web pages or repositories)?
I would like to investigate it.

bmunkhtulga, 2019-04-10:
Is there any advancement in this issue? I am facing exact same problem. But this time on the Ubuntu terminal it says 

> [ERROR] [1554892272.961044057]: Failed to write. (80000900)
> [ERROR] [1554892273.008740337]: Automatically change to normal mode.

On the other side, COBOTTA led turns non-blinking yellow and throws this error shown in the picture.

[image]

And exact same time, system had this error. `Buffer is empty.`

[image]

Thank you.

ToshitakaSuzuki, 2019-04-16:
@bmunkhtulga 
Sorry for my late reply.
Do you see this issue every time?

bmunkhtulga, 2019-04-17:
> @bmunkhtulga
> Sorry for my late reply.
> Do you see this issue every time?

Hopefully no. After changing these values. 

```
* The default value of "send_format": 0
* The default value of "recv_format": 2
```
Those errors happen only when COBOTTA is not able to reach destined position due to joint limit or something else.

RoboticsYY, 2019-09-05:
@bmunkhtulga @ToshitakaSuzuki This error happens pretty often on my side. I have checked the joint limits in the URDF file, and lowered them to much smaller than soft joint position limits. Actually this error happens in some very comfortable pose of COBOTTA.  Do you have other hints to fix this error? Thanks.

wzy19950426, 2020-09-03:
你好，请问那个Failed to write. (80000900)报错是怎么解决的？我在日志中也没有看到PC端的报错。只有ros这边运行到最后会显示这个错误。还请指教一下，谢谢。

TakahiroSawahiraki, 2020-09-10:
@wzy19950426 
Sorry. I don't understand chinese. So I reply in English.
Can you check the error log from [Setting] -> [Operation and Error Log] in your teach pendant?
Are there any other errors besides [80000900] at the same time?
Or are there an error dialog in the TP monitor when  [80000900] occurs?

crackls, 2023-06-23:
> Hi， my error code in the TP is 84201482, which reads "delay in generating the commands" (it is writen in Chinese) and my ROS throws out the ' failed to write(80000900)' error. how can i fix this? [...]

Did you mange to solve this problem. I am now working with Cobotta and yesterday everython was working but now it is not, and it is giving me this 84201482 error. It also gives me the error 83500121 that says SlvMove command was executed while Slave mode was not selected, even though I have selected Slave Mode. Thanks