DENSORobot/denso_robot_ros #14: How to change arm to gravity mode in ros

zhoubo2018, 2019-03-06:
I want to do a camera calibration to arm, but when I take example points of marker, I want to change the arm to gravity mode or teach mode, how can I do that? thanks.

ToshitakaSuzuki, 2019-03-11:
Sorry, denso_robot_bringup dose not be designed to allow gravity mode.

To consider another way, let me ask a question:
 While taking example points of maker, what you want to do is:
   - Move the arm by gravity mode.
   - Read the current joint angles.

Are these two?

Thank you.

zhoubo2018, 2019-03-11:
thanks for reply, yes, I want to the two things in ros,now I use move it to control the arm when camera take examples for calibration, but the result of calibration is not very good. because in techpad there is a direct mode that when  I hold the button on arm , cobotta can change to gravity mode. so I want to ask how to do that in ros.

ToshitakaSuzuki, 2019-04-02:
Sorry for my late reply.
COBOTTA cannot switch normal mode to direct mode automatically from ros.
It is necessary to use a teach pad and then push the button on arm for the safety reason.

Do you use some camera-robot calibration ros package or develop it on your own?

ToshitakaSuzuki, 2021-05-12:
I will close this issue due to no response.
If there is any progress, please let me know again.

closed 2021-05-12 by ToshitakaSuzuki