DENSORobot/denso_robot_ros #16: How to change the joint limit of cobotta arm

zhoubo2018, 2019-04-02:
I try to write moveit python file to control the arm, I write several points, sometimes the moveit plan is successful, but when the arm moves, it will reach the joint limit, the arm led will turn  yellow, I think it may be a joint limit setting in control box, so I want to ask you how to set that. Looking forward to your reply! Thank you very much.

ToshitakaSuzuki, 2019-04-02:
The joint angle limits are specified in URDF file.
But why are you trying to change the joint angle limits?
Can't you change your target points?

zhoubo2018, 2019-04-02:
yeah, because I use the windows/android app to move the arm to the position which I want to move in ros, it worked, so there might be some limit in ros or in b-cap

ToshitakaSuzuki, 2019-04-02:
> yeah, because I use the windows/android app to move the arm to the position which I want to move in ros

OK, I understand your requirement.

The problem might be that moveit generates trajectories that joint angle limits are not taken into account. But I don't why...

AlanSunAlan, 2019-04-25:
I think there might be necessary to change the joint limit in urdf. I found that the joint limits in the original urdf file are just too large. You can first determine the real limit value by moving the joints to the limit point by hand, then record all the values in order to change the urdf.