DENSORobot/denso_robot_ros #17: ROSConverter Issue

bmunkhtulga, 2019-04-04:
Thank you for providing ROS library for Denso robots.

I would like to control COBOTTA (standard type) using this package. In order to control robot other than VS060, we have to use ROSConverter. As I was following [this page's guide](http://wiki.ros.org/denso_robot_ros/ROSConverter), I encountered with this issue.

```
> tulga@tulga:~$ rosrun denso_robot_bringup install_robot_description.py /home/tulga/cobotta_description/
> Traceback (most recent call last):
>   File "/opt/ros/kinetic/lib/denso_robot_bringup/install_robot_description.py", line 97, in <module>
>     shutil.move(conf_dir, conf_pkg + "/config")
>   File "/usr/lib/python2.7/shutil.py", line 299, in move
>     copytree(src, real_dst, symlinks=True)
>   File "/usr/lib/python2.7/shutil.py", line 177, in copytree
>     os.makedirs(dst)
>   File "/usr/lib/python2.7/os.py", line 157, in makedirs
>     mkdir(name, mode)
> OSError: [Errno 13] Permission denied: '/opt/ros/kinetic/share/denso_robot_moveit_config/config/cobotta_config'
```
I tried giving python file all kinds of permission but had no luck. Also tried creating directory manually, but in python script, whenever there is same name directory, it calls `sys.exit()`. 

What should I do?

ToshitakaSuzuki, 2019-04-04:
Could you try to give the permission to the directory: '/opt/ros/kinetic/share/denso_robot_moveit_config/config'?

bmunkhtulga, 2019-04-04:
> Could you try to give the permission to the directory: '/opt/ros/kinetic/share/denso_robot_moveit_config/config'?

Thank you. I think I successfully loaded COBOTTA urdf and launch files. 

I had to give permission to some other folders such as `/opt/ros/kinetic/share/denso_robot_description/` `/opt/ros/kinetic/share/denso_robot_bringup/` other than `/opt/ros/kinetic/share/denso_robot_moveit_config/config`.

closed 2019-04-04 by bmunkhtulga

gavanderhoorn, 2019-04-04:
@ToshitakaSuzuki: is this an OK approach? Files and directories in `/opt` should not be writable by regular users. Those directories will also be overwritten by `apt` in the event of a package update.

ToshitakaSuzuki, 2019-04-04:
> @ToshitakaSuzuki: is this an OK approach? Files and directories in `/opt` should not be writable by regular users. Those directories will also be overwritten by `apt` in the event of a package update.

Thank you for pointing that out.
As you mentioned, this approach has some problems. 
But we would not like to prepare moveit config files and URDFs of all DENSO robots in denso_robot_ros package.
Because we have many kinds of robots.
Do you have any ideas?

gavanderhoorn, 2019-04-04:
> But we would not like to prepare moveit config files and URDFs of all DENSO robots in denso_robot_ros package.
> Because we have many kinds of robots.

Several OEM repositories for industrial robots in ROS-Industrial have adopted the approach of creating so called *robot support packages*. Instead of creating a `_description` package for each robot variant, or one (very) large single package containing *all* robot models and their variants, these robot support packages are a compromise: there is a single package per *robot model*, containing *all variants* of that series.

To give an example: an OEM (name him `mfg`) has a robot model X, with variants M and N. We then create a package: `mfg_x_support`. This package then contains the xacro macros (and meshes, launch files, etc) for variants `M` and `N`.

If there are 10 variants for one model, this leads to a single package for the model, instead of 10 (ie: one for each variant).

A little bit more rationale can be found [here](http://wiki.ros.org/Industrial/Tutorials/WorkingWithRosIndustrialRobotSupportPackages).

How useful this is depends of course on how the OEM has structured its product naming, but for many industrial robot manufacturers this works really well (see [fanuc_lrmate200ic_support](http://wiki.ros.org/fanuc_lrmate200ic_support) for the Fanuc LR Mate 200iC fi, which has 8 variants, but only a single support package).

You could consider adopting a similar approach.

gavanderhoorn, 2019-04-04:
As to MoveIt configurations: unfortunately these do not combine very well.

It is possible to parameterise them, so to allow the user to specify an additional argument on the command line when starting `demo.launch` (or a similar launch file), such as a `robot_type` argument (which can then be set to the name of one of the supported robots).

An alternative approach I've seen is using `xacro` to *generate* `.srdf` and other files when needed, again based on an input argument that defines the robot type (ie: robot model + specific variant) to generate for.

I'm not too much of a fan of either of these, because:

 1. reduced maintainability: MoveIt configurations are auto-generated by the MoveIt Setup Assistant (MSA), based on input from the user and a loaded robot model. Adding post-generation customisations to these auto-generated packages will complicate future updates, as the MSA doesn't know about these changes and will overwrite them. This requires the maintainer to port the changes he made to the new version of the configuration package, introducing maintenance overhead.
 1. utility of OEM provided MoveIt configs: MoveIt configurations of single robots, in isolation, without any kind of workcell or environment modelling are only useful for "testing" purposes. Any serious use of MoveIt with industrial robots will require adding environments, workpiece models, toolframes and end-effectors, which will add additional links and joints. This *requires* a re-run of the MSA by the user/integrator. If the default MoveIt package provided by the OEM is not going to be used too much anyway, customising it with tricks to support multiple robot variants doesn't make much sense to me.

Note: this is my personal opinion, not necessarily reflecting the ROS-Industrial consortium or community. Whether you (and/or DENSO) agree is something entirely up to you.

ToshitakaSuzuki, 2019-04-05:
@gavanderhoorn 
I appreciate your advice!
This seems very useful information for us.
DENSO WAVE just started supporting ROS officially last year.
So we do not know much about common ways in ROS and also ROS-Industrial.

Can I ask two questions?:
 1. Dose Fanuc officially support [Fanuc package](https://github.com/ros-industrial/fanuc)?
 2. What's the relationship between you and Fanuc? (Why do you maintain Fanuc package?)

gavanderhoorn, 2019-04-05:
@ToshitakaSuzuki: this starts to go off-topic for the problem the OP reported.

Perhaps we should take this discussion off-line.

Can you send me an email? You can find my address in any of the fanuc packages in the repository you linked.