DENSORobot/denso_robot_ros #18: execution slow

woshigeshabiaaa, 2019-04-08:
I am using this project with a denso robot.
I find that when i switcth to slave mode and use moveit to control the robot, it is visibilly much slower than i use the WincapsIII to execute some trajectories. 
I am using default velocity factors and set to 100% external speed. How can I speed up?

ToshitakaSuzuki, 2019-04-09:
Could you try to do the procedures in the following site?
http://wiki.ros.org/denso_robot_ros/AcquireVelAcc

woshigeshabiaaa, 2019-04-09:
> Could you try to do the procedures in the following site?
> http://wiki.ros.org/denso_robot_ros/AcquireVelAcc

yeah! it works! thanks!
you saved a lot of my times, sincerely gratitude for you! 
I once thought that if I used the URDF converter, it will autonomously export the right vel/acc limits. However it doesn't. I suggest modifying the tutorial to make  it explicitly two steps.

ToshitakaSuzuki, 2019-04-09:
>yeah! it works! thanks!
you saved a lot of my times, sincerely gratitude for you!

It's my pleasure!

>I suggest modifying the tutorial to make it explicitly two steps.

The explanation that parameters in URDF are dummy is already written in the tutorial (Please refer to "4.Setting speed/acceleration parameters").
http://wiki.ros.org/denso_robot_ros/Tutorials/How%20to%20control%20an%20RC8%20with%20MoveIt%21

closed 2019-04-09 by ToshitakaSuzuki