DENSORobot/denso_robot_ros #20: How to change the b_cap slave mode in ROS for cobotta

zhoubo2018, 2019-05-05:
Hello, after some testing the cobotta ros driver, we found sometimes the arm moveit will unexpected stop when using moveit (in remote app, there is some errors says "Slvmove instruction is executed in non-subbureau mode" and "instruction delay is generated"). the denso support ask me to change the b_cap slave mode in ros driver, so could you tell me where to change this parameter in this ros package? thanks a lot.

zhoubo2018, 2019-05-05:
I mean to change b_cap slave mode 0 to 1, just like that.

ToshitakaSuzuki, 2019-05-15:
Sorry for my late reply.

You can change slave mode by publishing the following topic:
 `rostopic pub /cobotta/ChangeMode std_msgs/Int32 "data: 2" `
"data:2" means changing to mode 2.

zhoubo2018, 2019-05-19:
yes, I have tried that, but it will first occur an error"Failed to write", and cobotta arm LED will become green, then I tried to type again, it will no error then the cobotta LED turn to white, is that right? Do I successfully change the mode?