DENSORobot/denso_robot_ros #21: Is there any way to call `bcap_service` from python script?

bmunkhtulga, 2019-05-10:
Is there any way to call `bcap_service` from python script? Like `rosrun denso_robot_bringup gripper.py`?

I have been trying to write client node in python, but since `bcap_service` server side was written in C++, i can't pass arguments to `variant[] vntArgs ` (there is no variant type in python, only has list type). 

Here is the code:

>         #create handler to call the service
>         bcap_service = rospy.ServiceProxy('bcap_service', bcap)
>
>         #define the request
>         func_id = 3
>         vt = 8
>         value_1 = 'b-CAP'
>         value_2 = 'CaoProv.DENSO.VRC'
>         value_3 = 'localhost'
>         value_4 = ''
>         vntArgs = [vt, value_1, vt, value_2, vt, value_3, vt, value_4]
> 
>         #call the service by passing the request to the handler
>         response = bcap_service(func_id, vntArgs)

_Originally posted by @bmunkhtulga in https://github.com/DENSORobot/denso_robot_ros/issues/13#issuecomment-490815292_

ToshitakaSuzuki, 2019-05-13:
variant type is defined as a message file in bcap_service package.
So the client node written in python will be like this:

```
import rospy
from bcap_service.msg import variant
from bcap_service.srv import *

def main():
    rospy.init_node('test_node', anonymous=True)
    bcapReq = bcap_service.srv.bcapRequest()
    bcapReq.func_id = 3
    bcapReq.vntArgs.append(variant(vt=8, value="b-CAP"))
    bcapReq.vntArgs.append(variant(vt=8, value="CaoProv.DENSO.VRC"))
    bcapReq.vntArgs.append(variant(vt=8, value="localhost"))
    bcapReq.vntArgs.append(variant(vt=8, value=""))

    bcapSrv = rospy.ServiceProxy('bcap_service', bcap)
    response = bcapSrv(bcapReq)

    #print(response.HRESULT)
    #print(response.vntRet.value)

if __name__ == '__main__':
    main()
```

This works in my environment.

bmunkhtulga, 2019-05-13:
Thank you. It is working in my environment too. I will close this as for now.

closed 2019-05-13 by bmunkhtulga