DENSORobot/denso_robot_ros #22: The simulation package is not working.

DongMuJi, 2019-05-15:
I have used the ROSConverter to make cobotta urdf. And use the follow command to launch cobotta simulation:
`roslaunch denso_robot_gazebo denso_robot_gazebo.launch robot_name:=cobotta`

I can  see the cobotta in gazebo,but on the teminal show some errors:

```
leo@leo-gaitech:~$ roslaunch denso_robot_gazebo denso_robot_gazebo.launch robot_name:=cobotta

... logging to /home/leo/.ros/log/bed5d050-7701-11e9-b3c6-3c95095e6245/roslaunch-leo-gaitech-27644.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://leo-gaitech:38943/

SUMMARY

PARAMETERS
 * /cobotta/arm_controller/joints: ['joint_1', 'join...
 * /cobotta/arm_controller/type: position_controll...
 * /cobotta/joint_state_controller/publish_rate: 125
 * /cobotta/joint_state_controller/type: joint_state_contr...
 * /robot_description: <?xml version="1....
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /use_gui: True
 * /use_sim_time: True

NODES
  /cobotta/
    controller_spawner (controller_manager/spawner)
  /
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    spawn_urdf (gazebo_ros/spawn_model)

auto-starting new master
process[master]: started with pid [27654]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to bed5d050-7701-11e9-b3c6-3c95095e6245
process[rosout-1]: started with pid [27667]
started core service [/rosout]
process[gazebo-2]: started with pid [27678]
process[gazebo_gui-3]: started with pid [27695]
process[spawn_urdf-4]: started with pid [27701]
process[cobotta/controller_spawner-5]: started with pid [27702]
process[robot_state_publisher-6]: started with pid [27703]
[ INFO] [1557918500.312192255]: Finished loading Gazebo ROS API Plugin.
[INFO] [1557918500.318659, 0.000000]: Controller Spawner: Waiting for service controller_manager/load_controller
[ INFO] [1557918500.320129046]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1557918500.388198388]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1557918500.388874684]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1557918500.849935635, 0.022000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1557918500.901576387, 0.069000000]: Physics dynamic reconfigure ready.
[ INFO] [1557918501.097208845, 0.252000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1557918501.280823201, 0.337000000]: Physics dynamic reconfigure ready.
[ INFO] [1557918501.326355838, 0.337000000]: Loading gazebo_ros_control plugin
[ERROR] [1557918501.326452032, 0.337000000]: GazeboRosControlPlugin missing <legacyModeNS> while using DefaultRobotHWSim, defaults to true.
This setting assumes you have an old package with an old implementation of DefaultRobotHWSim, where the robotNamespace is disregarded and absolute paths are used instead.
If you do not want to fix this issue in an old package just set <legacyModeNS> to true.

[ INFO] [1557918501.326555416, 0.337000000]: Starting gazebo_ros_control plugin in namespace: /cobotta
[ INFO] [1557918501.327339954, 0.337000000]: gazebo_ros_control plugin is waiting for model URDF in parameter [/robot_description] on the ROS param server.
[ WARN] [1557918501.440165801, 0.337000000]: Deprecated syntax, please prepend 'hardware_interface/' to 'PositionJointInterface' within the <hardwareInterface> tag in joint 'joint_1'.
[ERROR] [1557918501.442258641, 0.337000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/joint_1
[ WARN] [1557918501.442375819, 0.337000000]: Deprecated syntax, please prepend 'hardware_interface/' to 'PositionJointInterface' within the <hardwareInterface> tag in joint 'joint_2'.
[ERROR] [1557918501.444099104, 0.337000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/joint_2
[ WARN] [1557918501.444167319, 0.337000000]: Deprecated syntax, please prepend 'hardware_interface/' to 'PositionJointInterface' within the <hardwareInterface> tag in joint 'joint_3'.
[ERROR] [1557918501.445765501, 0.337000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/joint_3
[ WARN] [1557918501.445852412, 0.337000000]: Deprecated syntax, please prepend 'hardware_interface/' to 'PositionJointInterface' within the <hardwareInterface> tag in joint 'joint_4'.
[ERROR] [1557918501.447937253, 0.337000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/joint_4
[ WARN] [1557918501.448257265, 0.337000000]: Deprecated syntax, please prepend 'hardware_interface/' to 'PositionJointInterface' within the <hardwareInterface> tag in joint 'joint_5'.
[ERROR] [1557918501.452891279, 0.337000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/joint_5
[ WARN] [1557918501.452947484, 0.337000000]: Deprecated syntax, please prepend 'hardware_interface/' to 'PositionJointInterface' within the <hardwareInterface> tag in joint 'joint_6'.
[ERROR] [1557918501.454471424, 0.337000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/joint_6
[ INFO] [1557918501.462894933, 0.337000000]: Loaded gazebo_ros_control.
[ WARN] [1557918501.463366709, 0.338000000]: The default_robot_hw_sim plugin is using the Joint::SetPosition method without preserving the link velocity.
[ WARN] [1557918501.463422812, 0.338000000]: As a result, gravity will not be simulated correctly for your model.
[ WARN] [1557918501.463518558, 0.338000000]: Please set gazebo_pid parameters, switch to the VelocityJointInterface or EffortJointInterface, or upgrade to Gazebo 9.
[ WARN] [1557918501.463594439, 0.338000000]: For details, see https://github.com/ros-simulation/gazebo_ros_pkgs/issues/612
[spawn_urdf-4] process has finished cleanly
log file: /home/leo/.ros/log/bed5d050-7701-11e9-b3c6-3c95095e6245/spawn_urdf-4*.log
[INFO] [1557918501.531111, 0.404000]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1557918501.533042, 0.407000]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1557918501.534571, 0.408000]: Loading controller: joint_state_controller
[ERROR] [1557918501.537304059, 0.411000000]: Could not load controller 'joint_state_controller' because controller type 'joint_state_controller/JointStateController' does not exist.
[ERROR] [1557918501.537349003, 0.411000000]: Use 'rosservice call controller_manager/list_controller_types' to get the available types
[ERROR] [1557918502.538778, 1.390000]: Failed to load joint_state_controller
[INFO] [1557918502.539758, 1.391000]: Loading controller: arm_controller
[ERROR] [1557918502.542356776, 1.394000000]: Could not load controller 'arm_controller' because controller type 'position_controllers/JointTrajectoryController' does not exist.
[ERROR] [1557918502.542406101, 1.394000000]: Use 'rosservice call controller_manager/list_controller_types' to get the available types
[ERROR] [1557918503.544248, 2.349000]: Failed to load arm_controller
[INFO] [1557918503.546235, 2.349000]: Controller Spawner: Loaded controllers: 
[INFO] [1557918503.551513, 2.353000]: Started controllers: 
```

I try to use the rosservice call to see the controller_manager/list_controller_types
```
leo@leo-gaitech:~$ rosservice call /cobotta/controller_manager/list_controller_types "{}" 
types: []
base_classes: []
```

It shows no type. Now I want to use the gazebo to simulation Cobotta. Can you help me to solve this problem? Thank you very much.

ToshitakaSuzuki, 2019-05-15:
Is the joint state controller installed in your environment?

DongMuJi, 2019-05-16:
> Is the joint state controller installed in your environment?

Thank you for your reply. Now I have installed the join state controller package and try simulation again. Now the cobotta simulation is working. But when roslaunch the cobotta gazebo file. It shows some errors:
```
leo@leo-gaitech:~$ roslaunch denso_robot_gazebo denso_robot_gazebo.launch robot_name:=cobotta
... logging to /home/leo/.ros/log/dc6b179a-7780-11e9-b1b1-3c95095e6245/roslaunch-leo-gaitech-17387.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://leo-gaitech:36985/

SUMMARY
========

PARAMETERS
 * /cobotta/arm_controller/joints: ['joint_1', 'join...
 * /cobotta/arm_controller/type: position_controll...
 * /cobotta/joint_state_controller/publish_rate: 125
 * /cobotta/joint_state_controller/type: joint_state_contr...
 * /robot_description: <?xml version="1....
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /use_gui: True
 * /use_sim_time: True

NODES
  /cobotta/
    controller_spawner (controller_manager/spawner)
  /
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    spawn_urdf (gazebo_ros/spawn_model)

auto-starting new master
process[master]: started with pid [17397]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to dc6b179a-7780-11e9-b1b1-3c95095e6245
process[rosout-1]: started with pid [17410]
started core service [/rosout]
process[gazebo-2]: started with pid [17421]
process[gazebo_gui-3]: started with pid [17435]
process[spawn_urdf-4]: started with pid [17444]
process[cobotta/controller_spawner-5]: started with pid [17445]
process[robot_state_publisher-6]: started with pid [17446]
[ INFO] [1557973096.060341826]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1557973096.064772556]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[INFO] [1557973096.067114, 0.000000]: Controller Spawner: Waiting for service controller_manager/load_controller
[ INFO] [1557973096.128000208]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1557973096.129013249]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1557973096.600020553, 0.021000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1557973096.651371182, 0.067000000]: Physics dynamic reconfigure ready.
[ INFO] [1557973096.892352778, 0.143000000]: Loading gazebo_ros_control plugin
[ERROR] [1557973096.892447792, 0.143000000]: GazeboRosControlPlugin missing <legacyModeNS> while using DefaultRobotHWSim, defaults to true.
This setting assumes you have an old package with an old implementation of DefaultRobotHWSim, where the robotNamespace is disregarded and absolute paths are used instead.
If you do not want to fix this issue in an old package just set <legacyModeNS> to true.

[ INFO] [1557973096.892511311, 0.143000000]: Starting gazebo_ros_control plugin in namespace: /cobotta
[ INFO] [1557973096.893541132, 0.143000000]: gazebo_ros_control plugin is waiting for model URDF in parameter [/robot_description] on the ROS param server.
[ WARN] [1557973097.002245220, 0.143000000]: Deprecated syntax, please prepend 'hardware_interface/' to 'PositionJointInterface' within the <hardwareInterface> tag in joint 'joint_1'.
[ERROR] [1557973097.003697592, 0.143000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/joint_1
[ WARN] [1557973097.003809642, 0.143000000]: Deprecated syntax, please prepend 'hardware_interface/' to 'PositionJointInterface' within the <hardwareInterface> tag in joint 'joint_2'.
[ERROR] [1557973097.004865593, 0.143000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/joint_2
[ WARN] [1557973097.004957365, 0.143000000]: Deprecated syntax, please prepend 'hardware_interface/' to 'PositionJointInterface' within the <hardwareInterface> tag in joint 'joint_3'.
[ERROR] [1557973097.006052881, 0.143000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/joint_3
[ WARN] [1557973097.006143864, 0.143000000]: Deprecated syntax, please prepend 'hardware_interface/' to 'PositionJointInterface' within the <hardwareInterface> tag in joint 'joint_4'.
[ERROR] [1557973097.007188615, 0.143000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/joint_4
[ WARN] [1557973097.007255326, 0.143000000]: Deprecated syntax, please prepend 'hardware_interface/' to 'PositionJointInterface' within the <hardwareInterface> tag in joint 'joint_5'.
[ERROR] [1557973097.008252330, 0.143000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/joint_5
[ WARN] [1557973097.008295551, 0.143000000]: Deprecated syntax, please prepend 'hardware_interface/' to 'PositionJointInterface' within the <hardwareInterface> tag in joint 'joint_6'.
[ERROR] [1557973097.009299070, 0.143000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/joint_6
[ INFO] [1557973097.016336183, 0.143000000]: Loaded gazebo_ros_control.
[ WARN] [1557973097.016545161, 0.144000000]: The default_robot_hw_sim plugin is using the Joint::SetPosition method without preserving the link velocity.
[ WARN] [1557973097.016571412, 0.144000000]: As a result, gravity will not be simulated correctly for your model.
[... cut ...]
[INFO] [1557973097.276365, 0.401000]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1557973097.278136, 0.403000]: Loading controller: joint_state_controller
[INFO] [1557973097.287171, 0.412000]: Loading controller: arm_controller
[INFO] [1557973097.344256, 0.469000]: Controller Spawner: Loaded controllers: joint_state_controller, arm_controller
[INFO] [1557973097.347214, 0.472000]: Started controllers: joint_state_controller, arm_controller
```
Is it OK? or have solution to solve these errors?

Alibinwahid, 2019-07-09:
Hi I also tried the simulation today, it isn't working. It spawned the robot in gazebo environment but the controller doesn't seem to work well. following is the error I am getting:

> [ERROR] [1562659832.567592637, 0.510000000]: Could not find joint 'joint_1' in 'hardware_interface::PositionJointInterface'.
> [ERROR] [1562659832.567688997, 0.510000000]: Failed to initialize the controller
> [ERROR] [1562659832.567733803, 0.510000000]: Initializing controller 'arm_controller' failed
> [ERROR] [1562659833.569272, 1.510000]: Failed to load arm_controller 

Below is the complete output

> $ roslaunch denso_robot_gazebo denso_robot_gazebo.launch
> ... logging to /home/ali/.ros/log/041d268c-a221-11e9-9e1c-d46a6ad5baeb/roslaunch-ali-Inspiron-7572-12403.log
> Checking log directory for disk usage. This may take awhile.
> Press Ctrl-C to interrupt
> Done checking log file disk usage. Usage is <1GB.
> 
> started roslaunch server http://ali-Inspiron-7572:41333/
> 
> SUMMARY
> ========
> 
> PARAMETERS
>  * /robot_description: <robot name="vs06...
>  * /rosdistro: kinetic
>  * /rosversion: 1.12.14
>  * /use_gui: True
>  * /use_sim_time: True
>  * /vs060/arm_controller/joints: ['joint_1', 'join...
>  * /vs060/arm_controller/type: position_controll...
>  * /vs060/joint_state_controller/publish_rate: 125
>  * /vs060/joint_state_controller/type: joint_state_contr...
> 
> NODES
>   /vs060/
>     controller_spawner (controller_manager/spawner)
>   /
>     gazebo (gazebo_ros/gzserver)
>     gazebo_gui (gazebo_ros/gzclient)
>     robot_state_publisher (robot_state_publisher/robot_state_publisher)
>     spawn_urdf (gazebo_ros/spawn_model)
> 
> auto-starting new master
> process[master]: started with pid [12425]
> ROS_MASTER_URI=http://localhost:11311
> 
> setting /run_id to 041d268c-a221-11e9-9e1c-d46a6ad5baeb
> process[rosout-1]: started with pid [12449]
> started core service [/rosout]
> process[gazebo-2]: started with pid [12460]
> process[gazebo_gui-3]: started with pid [12476]
> process[spawn_urdf-4]: started with pid [12482]
> process[vs060/controller_spawner-5]: started with pid [12484]
> process[robot_state_publisher-6]: started with pid [12485]
> [ INFO] [1562659831.010957520]: Finished loading Gazebo ROS API Plugin.
> [ INFO] [1562659831.012624867]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
> [INFO] [1562659831.333492, 0.000000]: Controller Spawner: Waiting for service controller_manager/load_controller
> [ INFO] [1562659831.865085505, 0.021000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
> [ INFO] [1562659831.895376261, 0.051000000]: Physics dynamic reconfigure ready.
> [ INFO] [1562659832.285608967, 0.349000000]: Loading gazebo_ros_control plugin
> [ INFO] [1562659832.285823751, 0.349000000]: Starting gazebo_ros_control plugin in namespace: /vs060
> [ INFO] [1562659832.286612544, 0.349000000]: gazebo_ros_control plugin is waiting for model URDF in parameter [/robot_description] on the ROS param server.
> [ INFO] [1562659832.399804423, 0.349000000]: Joint mode switching is enabled
> [ INFO] [1562659832.408275963, 0.349000000]: Loaded gazebo_ros_control.
> [spawn_urdf-4] process has finished cleanly
> log file: /home/ali/.ros/log/041d268c-a221-11e9-9e1c-d46a6ad5baeb/spawn_urdf-4*.log
> [INFO] [1562659832.541414, 0.483000]: Controller Spawner: Waiting for service controller_manager/switch_controller
> [INFO] [1562659832.542482, 0.484000]: Controller Spawner: Waiting for service controller_manager/unload_controller
> [INFO] [1562659832.543450, 0.485000]: Loading controller: joint_state_controller
> [INFO] [1562659832.551161, 0.493000]: Loading controller: arm_controller
> [ERROR] [1562659832.567592637, 0.510000000]: Could not find joint 'joint_1' in 'hardware_interface::PositionJointInterface'.
> [ERROR] [1562659832.567688997, 0.510000000]: Failed to initialize the controller
> [ERROR] [1562659832.567733803, 0.510000000]: Initializing controller 'arm_controller' failed
> [ERROR] [1562659833.569272, 1.510000]: Failed to load arm_controller
> [INFO] [1562659833.569627, 1.511000]: Controller Spawner: Loaded controllers: joint_state_controller
> [INFO] [1562659833.573510, 1.515000]: Started controllers: joint_state_controller
[image]