DENSORobot/denso_robot_ros #23: Suggestion to fix bug in updating speed/acceleration parameters

Landau0323, 2019-06-22:
### What happened
I tried to update the speed limit of VS-068, following the procedure written in http://wiki.ros.org/denso_robot_ros/AcquireVelAcc .
After that, I tested whether the speed limit was correctly updated by doing 
`roslaunch denso_robot_bringup vs068_bringup.launch sim:=false ip_address:=192.168.0.1`
which resulted that, the robot didn't move fast, indicating that the speed limit was not updated.

Then I checked vs068_config/joint_limits.yaml, and found that spaces between colon and argument value are missing.
For example, 
`joint_1:
    has_velocity_limits:true
    max_velocity:3.92699081698724
    has_acceleration_limits:true
    max_acceleration:19.7335651876739`
should be corrected like
`joint_1:
    has_velocity_limits: true
    max_velocity: 3.92699081698724
    has_acceleration_limits: true
    max_acceleration: 19.7335651876739` 
After adding spaces in this file, the robot moved fast and the speed limit was correctly updated.

### What I suggest
Update update_joint_limits.py so that the spaces are correctly inserted when doing `rosrun denso_robot_bringup update_joint_limits.py vs068 ~/joint_limits.yaml`.

ToshitakaSuzuki, 2019-07-03:
Thank you for your suggestion.
We will fix update_joint_limits.py and also AcquireVelAcc as soon as possible.

ToshitakaSuzuki, 2019-07-12:
@Landau0323 
I updated!

Landau0323, 2019-07-30:
Thanks a lot!
Where can I get the new version of AcquireVelAcc?

DensoWaveRobot, 2019-07-30:
You can get the new version here.
https://www.denso-wave.com/en/robot/download/application/j1w3.html
Ver.3.58.0(07/08/2019) or later
http://wiki.ros.org/denso_robot_ros/AcquireVelAcc

closed 2019-08-09 by ToshitakaSuzuki