DENSORobot/denso_robot_ros #25: Catkin_make error(denso_robot_ros)

ryo-matsuzaka, 2019-07-13:
When I do 'catkin_make',  the following error happens.
Could you tell me the solution of it?
I already download all package file including 'bcap_core'.

[image]
[image]

DensoWaveRobot, 2019-07-15:
> When I do 'catkin_make', the following error happens.
> Could you tell me the solution of it?
> I already download all packages including 'bcap_core'.
What os do you use in your pc?
If you use windows, could you install in linux pc?

ryo-matsuzaka, 2019-07-16:
Thank you for your reply.
YES, I use windows OS.
So, you mean it doesn't work on ROSOnWindows?
If so, I will use linux.

DensoWaveRobot, 2019-07-16:
>So, you mean it doesn't work on ROSOnWindows?

yes.
When you've finished, please close this.

ryo-matsuzaka, 2019-07-16:
Ok, thank you very much.
I close this.

ryo-matsuzaka, 2019-07-17:
I Ireopend.
I challenged again. But I got same error.

[image]
[image]

DensoWaveRobot, 2019-07-17:
Could you see the CMakeOutput.log and CMakeError.log, as written your secound picture?
That's not error of denso_robot_ros.

gavanderhoorn, 2019-07-17:
@ryo-matsuzaka: did you copy the Catkin workspace that you used under Windows and now try to use it with Ubuntu?

If so: please remove `devel` and `build` folders first. Then try building the workspace.

Or are you using WSL?

ryo-matsuzaka, 2019-07-17:
@DensoWaveRobot 
Thank you for your reply. This is my error log.
[image]

ryo-matsuzaka, 2019-07-17:
@gavanderhoorn 
Thank you for asking. Yes, firstly I tried using ROSOnWindows. Secondary, I used WSL(Ubuntu18.04) and ROS melodic.
But I got the same error as I mentioned above.

gavanderhoorn, 2019-07-17:
> Thank you for asking. Yes, firstly I tried using ROSOnWindows. Secondary, I used WSL(Ubuntu18.04) and ROS melodic.
> But I got the same error as I mentioned above.

you cannot share (half-built) workspaces between all these three. The paths will be different and the tools also.

You'll have to make sure to remove the `build` and `devel` sub directories of your catkin workspace before trying to build it on another OS (and in this case: Windows, Ubuntu-in-WSL and Ubuntu-native are essentially different OS).

I'm not claiming that will solve your problem, but just wanted to make sure the no-sharing-workspaces was clear.

ryo-matsuzaka, 2019-07-17:
@gavanderhoor
Thirdly, I used WSL(Ubuntu16.04) and ROS kinetic. As a result,  I got another error as bellow.
[image]

gavanderhoorn, 2019-07-17:
WSL doesn't come with an X-server by default. So you cannot run GUI applications.

You'll have to install an X-server on Windows.

But even then WSL is going to not be the same experience as you would have with a native Ubuntu installation.

ryo-matsuzaka, 2019-07-17:
@gavanderhoorn 
Thank you for your kind advice and quick reply.  

Yes, I delete the 'build' and 'devel',  but I encountered the same experience between 'ROSOnWindows and WSL(Ubuntu 18.04). 
From now on I take care of naming workspaces.

> you cannot share (half-built) workspaces between all these three. The paths will be different and the tools also.
>
> You'll have to make sure to remove the `build` and `devel` sub directories of your [...]

Thank you for your kind advice and quick reply.  
Yes, I delete the 'build' and 'devel',  but I encountered the same experience between 'ROSOnWindows and WSL(Ubuntu 18.04). 
From now on I take care of naming workspaces.

ryo-matsuzaka, 2019-07-17:
> WSL doesn't come with an X-server by default. So you cannot run GUI applications.
> 
> You'll have to install an X-server on Windows.
> 
> But even then WSL is going to not be the same experience as you would have with a native Ubuntu installation.
> 
> Edit: note btw that this is not really a problem with the packages in this repository, so this seems to be off-topic for this tracker.
> 
> It's up to @DensoWaveRobot though.

Thank you for your advice.
I install an X-server on Windows. 
Ok, I am going to close this tracker and might set another tracker.
Anyway, thank you so much.

7675t, 2019-07-18:
Generally speaking, ROS on windows cause troubles in many different ways. I recommend to stick to Ubuntu unless you want to have fun of troubleshoot 😄

ryo-matsuzaka, 2019-07-18:
@7675t 
> Generally speaking, ROS on windows cause troubles in many different ways. I recommend to stick to Ubuntu unless you want to have fun of troubleshoot 😄

Thank you so much for your advice. Yes, I don't like trouble shooting. So I will use linux PC as you recommend.

Thus, I close this tracker.

ryo-matsuzaka, 2019-07-18:
Dear people who gave me any advice,

I could solve problems mentioned above.
As gavanderhoorn mentioned, I installed X server.
Then, I was able to launch rviz, gazebo, denso robot package under environment of WSL(Ubuntu16.04) and ROS kinetic.

Thank you.

ryo-matsuzaka, 2019-07-18:
I close it.

closed 2019-07-18 by ryo-matsuzaka