DENSORobot/denso_robot_ros #26: OpenAI ROS tutorial error/TypeError: __init__() got an unexpected keyword argument 'timestep_limit'

ryo-matsuzaka, 2019-07-20:
By following the first video tutorial at http://wiki.ros.org/openai_ros, I did 'roslaunch turtle2_openai_ros_example start_training_maze.launch' .
Then, I got 'TypeError: __init__() got an unexpected keyword argument 'timestep_limit''. Could you tell me how should I cope with?
This is the detail of my command line.

////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////

NODES
  /
    example_turtlebot2_maze_qlearn (turtle2_openai_ros_example/my_start_qlearning_maze.py)

ROS_MASTER_URI=http://localhost:11311

process[example_turtlebot2_maze_qlearn-1]: started with pid [5828]
Traceback (most recent call last):
  File "/home/usr/catkin_ws/src/openai_examples_projects/turtle2_openai_ros_example/scripts/my_start_qlearning_maze.py", line 12, in <module>
    import my_turtlebot2_maze
  File "/home/usr/catkin_ws/src/openai_examples_projects/turtle2_openai_ros_example/scripts/my_turtlebot2_maze.py", line 13, in <module>
    timestep_limit=timestep_limit_per_episode,
  File "/usr/local/lib/python2.7/dist-packages/gym/envs/registration.py", line 153, in register
    return registry.register(id, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/gym/envs/registration.py", line 147, in register
    self.env_specs[id] = EnvSpec(id, **kwargs)
TypeError: __init__() got an unexpected keyword argument 'timestep_limit'
[example_turtlebot2_maze_qlearn-1] process has died [pid 5828, exit code 1, cmd /home/usr/catkin_ws/src/openai_examples_projects/turtle2_openai_ros_example/scripts/my_start_qlearning_maze.py __name:=example_turtlebot2_maze_qlearn __log:=/home/usr/.ros/log/aa25e98c-aa88-11e9-9178-bcc342af79b5/example_turtlebot2_maze_qlearn-1.log].
log file: /home/usr/.ros/log/aa25e98c-aa88-11e9-9178-bcc342af79b5/example_turtlebot2_maze_qlearn-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

> Blockquote

ToshitakaSuzuki, 2019-07-22:
Your question is not related to this repository.
Following pages might help you:
https://github.com/openai/gym/issues/445
https://answers.ros.org/question/325551/openai_ros-openai_examples_projects-q-learning-example-issue/
https://stackoverflow.com/questions/42126772/typeerror-init-got-an-unexpected-keyword-argument-method/51660035

ryo-matsuzaka, 2019-07-22:
Thank you so much.
I check it right now.

closed 2019-07-22 by ryo-matsuzaka