DENSORobot/denso_robot_ros #27: Unable to find package: controller_interface / Melodic compatibility

tswie, 2019-07-22:
Hello,

I'm currently trying to control a VS6577 robot using this package on ROS Melodic. So far, I am capable of running my project in Gazebo and the robot responds to commands issued to it through RViz. Upon trying to run the project live, I get this issue each time: 

```
[INFO] [1563483607.986431]: Controller Spawner: Waiting for service controller_manager/load_controller
[ INFO] [1563483608.501582364]: Stereo is NOT SUPPORTED
[ INFO] [1563483608.502584956]: OpenGl version: 4.6 (GLSL 4.6).
Succeeded in initializing the robot
terminate called after throwing an instance of 'pluginlib::ClassLoaderException'
  what():  Unable to find package: controller_interface
[vs6577/denso_robot_control-3] process has died [pid 7375, exit code -6, cmd /home/cerlabdesktop/depowdering/devel/lib/denso_robot_control/denso_robot_control __name:=denso_robot_control __log:=/home/cerlabdesktop/.ros/log/05037648-a99f-11e9-b1c8-509a4c1bf48c/vs6577-denso_robot_control-3.log].
log file: /home/cerlabdesktop/.ros/log/05037648-a99f-11e9-b1c8-509a4c1bf48c/vs6577-denso_robot_control-3*.log
[ INFO] [1563483611.815198199]: Loading robot model 'vs6577'...
[ WARN] [1563483611.847747450]: Kinematics solver doesn't support #attempts anymore, but only a timeout.
Please remove the parameter '/rviz_cerlabdesktop_7343_8363000961692927200/arm/kinematics_solver_attempts' from your configuration.
[ INFO] [1563483611.946172416]: Starting planning scene monitor
[ INFO] [1563483611.948721503]: Listening to '/move_group/monitored_planning_scene'
[ INFO] [1563483611.949409933]: waitForService: Service [/get_planning_scene] has not been advertised, waiting...
[ WARN] [1563483612.922922069]: Waiting for vs6577/arm_controller/follow_joint_trajectory to come up
[ INFO] [1563483616.968889332]: Failed to call service get_planning_scene, have you launched move_group? at /tmp/binarydeb/ros-melodic-moveit-ros-planning-1.0.1/planning_scene_monitor/src/planning_scene_monitor.cpp:495
[ INFO] [1563483617.022663224]: Constructing new MoveGroup connection for group 'arm' in namespace ''
[ WARN] [1563483618.923391565]: Waiting for vs6577/arm_controller/follow_joint_trajectory to come up
[ERROR] [1563483624.923596247]: Action client not connected: vs6577/arm_controller/follow_joint_trajectory
[ INFO] [1563483624.937697712]: Returned 0 controllers in list
[ INFO] [1563483624.946942656]: Trajectory execution is managing controllers

```

 After getting this issue the first time, the error message switches to: 

```
[ INFO] [1563817852.069669255]: MoveGroup debug mode is OFF
Starting planning scene monitors...
[ INFO] [1563817852.069702949]: Starting planning scene monitor
[ INFO] [1563817852.070955503]: Listening to '/planning_scene'
[ INFO] [1563817852.070974178]: Starting world geometry update monitor for collision objects, attached objects, octomap updates.
[ INFO] [1563817852.072195642]: Listening to '/collision_object' using message notifier with target frame 'world '
[ INFO] [1563817852.073332544]: Listening to '/planning_scene_world' for planning scene world geometry
[ERROR] [1563817852.085463921]: Failed to connect real controller. (83501032)
[ INFO] [1563817852.090143339]: Listening to '/attached_collision_object' for attached collision objects
Planning scene monitors started.
[ INFO] [1563817852.109059295]: Initializing OMPL interface using ROS parameters
[ INFO] [1563817852.123005684]: Using planning interface 'OMPL'
[ INFO] [1563817852.126260570]: Param 'default_workspace_bounds' was not set. Using default value: 10
[ INFO] [1563817852.126581205]: Param 'start_state_max_bounds_error' was set to 0.1
[ INFO] [1563817852.126824031]: Param 'start_state_max_dt' was not set. Using default value: 0.5
[ INFO] [1563817852.127085783]: Param 'start_state_max_dt' was not set. Using default value: 0.5
[ INFO] [1563817852.127306797]: Param 'jiggle_fraction' was set to 0.05
[ INFO] [1563817852.127531568]: Param 'max_sampling_attempts' was not set. Using default value: 100
[ INFO] [1563817852.127573359]: Using planning request adapter 'Add Time Parameterization'
[ INFO] [1563817852.127593317]: Using planning request adapter 'Fix Workspace Bounds'
[ INFO] [1563817852.127622302]: Using planning request adapter 'Fix Start State Bounds'
[ INFO] [1563817852.127648352]: Using planning request adapter 'Fix Start State In Collision'
[ INFO] [1563817852.127672739]: Using planning request adapter 'Fix Start State Path Constraints'
[FATAL] [1563817852.129933186]: Exception while loading controller manager 'moveit_simple_controller_manager/MoveItSimpleControllerManager': According to the loaded plugin descriptions the class moveit_simple_controller_manager/MoveItSimpleControllerManager with base class type moveit_controller_manager::MoveItControllerManager does not exist. Declared types are  moveit_controller_manager_example/MoveItExampleControllerManager moveit_fake_controller_manager/MoveItFakeControllerManager
[ INFO] [1563817852.137131077]: Trajectory execution is managing controllers

```

At this point, I need to restart the robot controller in order to return to the first error message (controller_interface not found). I've tried uninstalling ros-melodic-controllers and ros_control and recompiling them as part of the project. ROS itself can find the controller_interface package; it is also listed as required in the package.xml and CMakeLists.txt. The project compiles as well, it just fails to run in non-simulation mode.

Is this a known Melodic issue? The project otherwise works in Melodic so far, and I would prefer to keep using our 18.04 setup if possible.

ToshitakaSuzuki, 2019-07-25:
Thank you for your reporting.
I get the same error as yours in ubuntu 18.04 / Melodic.
I'm still investigating the cause of this error.
> terminate called after throwing an instance of 'pluginlib::ClassLoaderException
>  what():  Unable to find package: controller_interface`

The exception throws from pluginlib packages:
https://github.com/ros/pluginlib/blob/54c925eebf8fd8db0c5e7e9ea18cba1d32402909/pluginlib/include/pluginlib/class_loader_imp.hpp#L85-L87

As you said, although rospack can find the controller_interface, `ros::package::getPath(package_)` fails to find `package_`("controller_interface") and then returns empty string.
To be sure getPath() works correctly,  I wrote the minimal test code like this;
`string path = ros::package::getPath("controller_interface");`
It works and returns the correct package path...

tswie, 2019-08-02:
Hello,

Just for clarification, is it just the returned path that is correct or does the robot itself actually respond when explicitly asking for the path? If the former, is there any update on a fix for the issue?

tswie, 2019-08-19:
Hello,

Is there any update on this issue?

haraisao, 2020-02-21:
Hello guys,

I found the cause of this issue.
The problem is caused by the tinyxml2 library that included in a denso_robot_core package.
That library is not support 'xml-model' predeclare tag in a package.xml, so you can solve this problem with modified tinyxml2.c and tinyxml2.h as followings:

diff --git a/denso_robot_core/include/denso_robot_core/tinyxml2.h b/denso_robot_core/include/denso_robot_core/tinyxml2.h
index d932ff2..768685f 100644
--- a/denso_robot_core/include/denso_robot_core/tinyxml2.h
+++ b/denso_robot_core/include/denso_robot_core/tinyxml2.h
@@ -871,6 +871,7 @@ protected:
     XMLDocument*	_document;
     XMLNode*		_parent;
     mutable StrPair	_value;
+    int                 _parseLineNum;
 
     XMLNode*		_firstChild;
     XMLNode*		_lastChild;
diff --git a/denso_robot_core/src/tinyxml2.cpp b/denso_robot_core/src/tinyxml2.cpp
index ca98b64..82b4a4f 100644
--- a/denso_robot_core/src/tinyxml2.cpp
+++ b/denso_robot_core/src/tinyxml2.cpp
@@ -952,6 +952,8 @@ char* XMLNode::ParseDeep( char* p, StrPair* parentEnd )
             break;
         }
 
+	int initialLineNum = node->_parseLineNum;
+
         StrPair endTag;
         p = node->ParseDeep( p, &endTag );
         if ( !p ) {
@@ -966,11 +968,31 @@ char* XMLNode::ParseDeep( char* p, StrPair* parentEnd )
         if ( decl ) {
                 // A declaration can only be the first child of a document.
                 // Set error, if document already has children.
+#if 0
                 if ( !_document->NoChildren() ) {
                         _document->SetError( XML_ERROR_PARSING_DECLARATION, decl->Value(), 0);
                         DeleteNode( decl );
                         break;
                 }
+#else
+            // Declarations are only allowed at document level
+            bool wellLocated = ( ToDocument() != 0 );
+            if ( wellLocated ) {
+                // Multiple declarations are allowed but all declarations
+                // must occur before anything else
+                for ( const XMLNode* existingNode = _document->FirstChild(); existingNode; existingNode = existingNode->NextSibling() ) {
+                    if ( !existingNode->ToDeclaration() ) {
+                        wellLocated = false;
+                        break;
+                    }
+                }
+            }
+            if ( !wellLocated ) {
+                _document->SetError( XML_ERROR_PARSING_DECLARATION, 0, 0);
+                DeleteNode( node );
+                break;
+            }
+#endif
         }
 
         XMLElement* ele = node->ToElement();

DensoWaveRobot, 2020-02-25:
@haraisao 
Thank you for teaching me so kindly.
We will test it and update it.
Thank you so much!

Shunichi09, 2020-10-07:
Hi I think I also got the same error with Cobotta.
Did anyone solve this ??

`terminate called after throwing an instance of 'pluginlib::ClassLoaderException'
  what():  Unable to find package: controller_interface
[cobotta/denso_robot_control-3] process has died [pid 856, exit code -6, cmd /home/catkin_ws/devel/lib/denso_robot_control/denso_robot_control __name:=denso_robot_control __log:=/root/.ros/log/372726a8-08ab-11eb-95f7-0242ac110002/cobotta-denso_robot_control-3.log].
log file: /root/.ros/log/372726a8-08ab-11eb-95f7-0242ac110002/cobotta-denso_robot_control-3*.log`

DensoWaveRobot, 2020-10-07:
@Shunichi09 
Have you tried the modification that haraiso showed? 
https://github.com/DENSORobot/denso_robot_ros/issues/27#issuecomment-589495455

I have confirmed that this modify  fix the error you posted.
The official integrate to ROS melodic is in preparation. 
Sorry. Please wait for a while.

Shunichi09, 2020-10-07:
Thank you for the comments!!
I tried the modification. First, it did not work but I rebooted everything including the cobotta. Then, it worked!!
Thanks, again.