DENSORobot/denso_robot_ros #28: Could not find joint 'joint_1' in 'hardware_interface::PositionJointInterface'

roslearner, 2019-07-25:
when i run $ roslaunch denso_robot_gazebo denso_robot_gazebo.launch robot_name:=cobotta
there are some errors: [ERROR] : Could not find joint 'joint_1' in 'hardware_interface::PositionJointInterface'.
[ERROR] : Failed to initialize the controller
[ERROR] : Initializing controller 'arm_controller' failed libGL error: failed to create drawable
[ERROR]: Failed to load arm_controller 
if someone can solve it ,thank you very much.

ToshitakaSuzuki, 2019-07-25:
Can you describe more detailed environment such as OS and ROS versions ?

Also please try to run the launch file without `robot_name:=cobotta` argument as follows:
`$ roslaunch denso_robot_gazebo denso_robot_gazebo.launch`
 I expect it should work with default robot setting (vs060).

roslearner, 2019-07-25:
thank you for your reply.my computer is ubuntu16.04 and ros-kinetic,and i have installed your packages.when i try to run $ roslaunch denso_robot_gazebo denso_robot_gazebo.launch，three is a warning:[WARN] : Controller Spawner couldn't find the expected controller_manager ROS interface,and the robot arm is sleeping down gradually in the map,just like this:
[image]

ToshitakaSuzuki, 2019-07-25:
Make sure you have installed all required packages.
`$ sudo apt-get install ros-kinetic-ros-control ros-kinetic-ros-controllers ros-kinetic-gazebo-ros-control `

roslearner, 2019-07-26:
i have tried to install above packages you said,and run $ roslaunch denso_robot_gazebo denso_robot_gazebo.launch or run $ roslaunch denso_robot_gazebo denso_robot_gazebo.launch robot_name:=cobotta , then error occurs as below:
[image]

ToshitakaSuzuki, 2019-07-26:
Did you modify URDF files from the original one?
I assume your urdf file is missing `<hardwareInterface>` tag or  dose not describe `PositionJointInterface`.

roslearner, 2019-07-26:
i download "master" version of your package,i donot modify URDF files,the URDF files is below:
[image]

roslearner, 2019-07-26:
<hardwareInterface> tag described in the denso_robot_control.yaml  is below:
[image]

ToshitakaSuzuki, 2019-07-26:
OK, then try to install another dependencies:
`$ sudo apt install ros-kinetic-controller-manager 
	ros-kinetic-position-controllers 
	ros-kinetic-joint-state-controller 
	ros-kinetic-joint-trajectory-controller 
	ros-kinetic-joint-limits-interface 
	ros-kinetic-transmission-interface
`

roslearner, 2019-07-26:
now it can stand,maybe i can control it,can you tell me which topic i can use to control the robot or how can i control it,thank you.
[image]
[image]

roslearner, 2019-07-26:
How can i control the robot'motion in Gazebo simulation,is there any reference documents?if there has,please tell me ,thank you very much.

ToshitakaSuzuki, 2019-07-26:
@roslearner 
> now it can stand,maybe i can control it,can you tell me which topic i can use to control the robot or how can i control it,thank you.

How did you solve it?
Please share your knowledge.

To control the robot from topic, you can use `/cobotta/arm_controller/follow_joint_trajecotry`.

ToshitakaSuzuki, 2019-07-26:
And please close the issue if you solve the problem.

roslearner, 2019-07-27:
maybe just install the packages you have said and update,and run run $ roslaunch denso_robot_gazebo denso_robot_gazebo.launch robot_name:=cobotta.i will try to control the robot next Monday,i donot know whether error will occur .when i can control the robot by topic,i will close the issue.thank you very much.

roslearner, 2019-07-29:
@ToshitakaSuzuki ,hello,i want to control cobotta by topic :$ rostopic pub /cobotta/arm_controller/command trajectory_msgs/JointTrajectory
[image]  
but error occurs as blew:Cannot create trajectory from message. It does not contain the expected joints.
Would you tell me whether it is wrong and how to publish topic to control. if it is convenient,please tell me the complete topic command .thank you

ToshitakaSuzuki, 2019-07-29:
@roslearner 
As the error says, you should specify joint_names.
This will help you to understand  JointTrajectory message.
https://groups.google.com/forum/#!topic/moveit-users/vBtNgsqHx7o

Btw, what your purpose of control robot by topic?
Do you want to just move some point to point, or to move robot by your original trajectory?

roslearner, 2019-07-29:
now i just try to control by topic,and then i want to control the robot in gazebo randomly by writing program,move from one point to another point,pick and place and so on.

roslearner, 2019-07-30:
@ToshitakaSuzuki  when i run $ roslaunch denso_robot_gazebo denso_robot_gazebo.launch robot_name:=cobotta,and then run $ rostopic pub /cobotta/arm_controller/command trajectory_gs/JointTrajectory "header:
  seq: 0
  stamp:
    secs: 0
    nsecs: 0
  frame_id: ''
joint_names:
- '["joint_1","joint_2","joint_3"]'
points:
- positions: [0,0,2.0]
  velocities: [-1.0,1.69,2.0]
  accelerations: [1,2,1]
  effort: [0]
  time_from_start: {secs: 0, nsecs: 0}" 
There occurs an error: Cannot create trajectory from message. It does not contain the expected joints,and the cobotta doesnot move.do you know how the error is and how to solve it??
[image]