DENSORobot/denso_robot_ros #29: Failed to change to slave mode. (83501024) for cobotta

RoboticsYY, 2019-08-28:
Hi, I'm ready to launch `roslaunch denso_robot_control denso_robot_control.launch`, but with the urdf files from `denso_cobotta_ros`. I also add a `cobotta.launch.xml` as the same way of VS060:
```
<launch>
	<param name="robot_name" value="CVR038A1-NV6-NNN-NNN"/>
	<param name="robot_joints" value="6"/>
	<param name="joint_1" value="1"/>
	<param name="joint_2" value="1"/>
	<param name="joint_3" value="1"/>
	<param name="joint_4" value="1"/>
	<param name="joint_5" value="1"/>
	<param name="joint_6" value="1"/>
	<param name="arm_group" value="1"/>
</launch>
``` 
After fixing the internet token executable problem, I can see the motor can be powered on, but I got the error`[ERROR] [ros.denso_robot_control]: Failed to change to slave mode. (83501024)`:

The entire outputs are:
```
SUMMARY
========

PARAMETERS
 * /cobotta/arm_controller/joints: ['joint_1', 'join...
 * /cobotta/arm_controller/type: position_controll...
 * /cobotta/arm_group: 1
 * /cobotta/config_file: /home/yy/ws_denso...
 * /cobotta/conn_type: tcp
 * /cobotta/controller_name: 
 * /cobotta/controller_type: 8
 * /cobotta/invoke_timeout: 180000
 * /cobotta/ip_address: 192.168.0.10
 * /cobotta/joint_1: 1
 * /cobotta/joint_2: 1
 * /cobotta/joint_3: 1
 * /cobotta/joint_4: 1
 * /cobotta/joint_5: 1
 * /cobotta/joint_6: 1
 * /cobotta/joint_state_controller/publish_rate: 125
 * /cobotta/joint_state_controller/type: joint_state_contr...
 * /cobotta/port_number: 5007
 * /cobotta/recv_format: 292
 * /cobotta/retry_count: 5
 * /cobotta/robot_joints: 6
 * /cobotta/robot_name: CVR038A1-NV6-NNN-NNN
 * /cobotta/send_format: 288
 * /cobotta/timeout: 3000
 * /cobotta/wait_time: 0
 * /cobotta/watchdog_timer: 400
 * /robot_description: <?xml version="1....
 * /rosdistro: melodic
 * /rosversion: 1.14.3

NODES
  /cobotta/
    controller_spawner (controller_manager/spawner)
    denso_robot_control (denso_robot_control/denso_robot_control)
  /
    robot_state_publisher (robot_state_publisher/robot_state_publisher)

auto-starting new master
process[master]: started with pid [23949]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 946aa6e2-c98b-11e9-8860-d4bed94f78ea
process[rosout-1]: started with pid [23960]
started core service [/rosout]
process[cobotta/denso_robot_control-2]: started with pid [23968]
process[cobotta/controller_spawner-3]: started with pid [23969]
process[robot_state_publisher-4]: started with pid [23970]
[INFO] [rosout]: Controller Spawner: Waiting for service controller_manager/load_controller
[ERROR] [ros.denso_robot_control]: Failed to change to slave mode. (83501024)
[cobotta/denso_robot_control-2] process has finished cleanly
log file: /home/yy/.ros/log/946aa6e2-c98b-11e9-8860-d4bed94f78ea/cobotta-denso_robot_control-2*.log
[WARN] [rosout]: Controller Spawner couldn't find the expected controller_manager ROS interface.
[cobotta/controller_spawner-3] process has finished cleanly
log file: /home/yy/.ros/log/946aa6e2-c98b-11e9-8860-d4bed94f78ea/cobotta-controller_spawner-3*.log
```
The error log of remote TP shows that this may happen if two programs take the control of robot at the same time. How can I fix this error?

RoboticsYY, 2019-08-29:
After rebooting cobotta, I found this happens because I run into this [issue](https://github.com/DENSORobot/denso_robot_ros/issues/27) at the first time when I tried to launch the control program. Then if I launch the control program again, seems the previous failure run did not exit successfully and gave up the control of the robot. But cobotta is still thinking the previous program is taking control.

luuquan, 2019-12-10:
Have you been able to deal with this problem? I have got the same issues on VP6242.

Environment:
Ubuntu 18.04 + ROS Melodic

RoboticsYY, 2019-12-10:
@luuquan I recommend to use OSS mode, which is much more stable.

DensoWaveRobot, 2019-12-10:
@luuquan 
Sorry. 
This package supports Indigo and kinetic.
We will support Melodic in Next year.

tswie, 2019-12-14:
> @luuquan
> Sorry.
> This package supports Indigo and kinetic.
> We will support Melodic in Next year.

Thank you for addressing the issue!

Do you have a rough ETA? Our project relies on the DENSO arm and we'd like to be able to use Melodic. Also, I'm assuming this would be for the overall denso_robot_ros package, correct?

haraisao, 2020-02-21:
Hello,

When you will reset the cobotta using moveit, you need cancel a current planning motion and  publish  'ChangeMode' message towice.
To cancel a current planning motion, the 'Start State' set to <current> and 'Goal State' set to <same as start> by using the Planning tab in MotionPlanning in rviz.
Then you should press 'Plan' button and ' Execute' button.
After that, publishing 'ChangeMode' message with value as '0x202' will reset the current error, the  the led will turn to white.
Then publishing 'ChangeMode' message with value as '0x202' will the cobotta turn to bCap slave mode.

yo4hi6o, 2022-02-15:
Hello,

Currently, the same error occurs in my environment and roslaunch does not start.
Do you know what caused this error?

Environment

VS050A3-AV6-NNN-NNN
RC8-VSA-NNNM-NN-NN　　      Ver 1.5.8

ROS            Melodic
GitBranch    Master

1.Create a description file for VS050A3 using the ROS converter.

2.ROS Tool　I tried to get the speed information using Acquire VelAcc, but an error occurred and I could not get it. Therefore, copy the joint limits file of the VS060 file on Github.

Operating procedure

`roslaunch denso_robot_bringup vs050a3_bringup.launch sim:=false ip_address:=192.168.0.2`

Execution status

```
yo4hi6o@yo4hi6o:~/denso_ws$ roslaunch denso_robot_bringup vs050a3_bringup.launch sim:=false ip_address:=192.168.0.2
... logging to /home/yo4hi6o/.ros/log/fba5e768-8e14-11ec-993e-54ab3a5cd6b4/roslaunch-yo4hi6o-15450.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://yo4hi6o:44025/

SUMMARY
========

PARAMETERS
 * /joint_state_publisher/publish_frequency: 125.0
 * /joint_state_publisher/source_list: ['/vs050a3/joint_...
 * /joint_state_publisher/use_gui: False
 * /move_group/allow_trajectory_execution: True
 * /move_group/arm/longest_valid_segment_fraction: 0.05
 * /move_group/arm/planner_configs: ['SBLkConfigDefau...
 * /move_group/arm/projection_evaluator: joints(joint_1,jo...
 * /move_group/capabilities: move_group/MoveGr...
 * /move_group/controller_list: [{'default': True...
 * /move_group/controller_manager_ns: controller_manager
 * /move_group/jiggle_fraction: 0.05
 * /move_group/max_range: 5.0
 * /move_group/max_safe_path_cost: 1
 * /move_group/moveit_controller_manager: moveit_simple_con...
 * /move_group/moveit_manage_controllers: True
 * /move_group/octomap_resolution: 0.025
 * /move_group/planner_configs/BKPIECEkConfigDefault/type: geometric::BKPIECE
 * /move_group/planner_configs/ESTkConfigDefault/type: geometric::EST
 * /move_group/planner_configs/KPIECEkConfigDefault/type: geometric::KPIECE
 * /move_group/planner_configs/LBKPIECEkConfigDefault/type: geometric::LBKPIECE
 * /move_group/planner_configs/PRMkConfigDefault/type: geometric::PRM
 * /move_group/planner_configs/PRMstarkConfigDefault/type: geometric::PRMstar
 * /move_group/planner_configs/RRTConnectkConfigDefault/type: geometric::RRTCon...
 * /move_group/planner_configs/RRTkConfigDefault/type: geometric::RRT
 * /move_group/planner_configs/RRTstarkConfigDefault/type: geometric::RRTstar
 * /move_group/planner_configs/SBLkConfigDefault/type: geometric::SBL
 * /move_group/planner_configs/TRRTkConfigDefault/type: geometric::TRRT
 * /move_group/planning_plugin: ompl_interface/OM...
 * /move_group/planning_scene_monitor/publish_geometry_updates: True
 * /move_group/planning_scene_monitor/publish_planning_scene: True
 * /move_group/planning_scene_monitor/publish_state_updates: True
 * /move_group/planning_scene_monitor/publish_transforms_updates: True
 * /move_group/request_adapters: default_planner_r...
 * /move_group/sensors: []
[... 163 lines cut ...]
*     - QueryPlannersService
*     - StateValidationService
********************************************************

[ INFO] [1644898149.929385283]: MoveGroup context using planning plugin ompl_interface/OMPLPlanner
[ INFO] [1644898149.929424795]: MoveGroup context initialization complete

You can start planning now!

[ INFO] [1644898151.005178974]: Ready to take commands for planning group arm.
[ INFO] [1644898151.005314032]: Looking around: no
[ INFO] [1644898151.005371093]: Replanning: no
^C[rviz_yo4hi6o_15450_5623600324430190867-7] killing on exit
[move_group-6] killing on exit
[robot_state_publisher-5] killing on exit
[vs050a3/controller_spawner-4] killing on exit
[joint_state_publisher-2] killing on exit
[WARN] [1644898158.184648]: Controller Spawner couldn't find the expected controller_manager ROS interface.
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```