DENSORobot/denso_robot_ros #34: How do I launch Pac Scripts on the Cobotta using Ros?

zeldamax, 2020-11-30:
I have created a few pac scripts on my Cobotta (non OSS type). The scripts are located on the Cobotta. In the past, I launched these scripts using Virtual TP, but is it also possible to launch them using ros? I have already changed the executable token to Ethernet. I would imagine there is a way to start a specific pac script by its name, could that be and if so how? Thank you in advance.

DensoWaveRobot, 2020-12-04:
@zeldamax 
I'm sorry for the rate reply.

You can execute pac script with bcapservice node.
This is overview how to use bcapservice and executescript named "pro1.pcs".

1.Do the follows
ubuntu@roslaunch denso_robot_bringup sim:=false ip_address:=<ROBOT's IP Address>
2.Do the follows on new terminal
ubuntu@roslaunch bcap_service bcap_service.launch ip_address:=<ROBOT's IP Address>
3.Do the follows on new terminal
ubuntu@rosservice call /bcap_service '{func_id: 3, vntArgs: [{vt: 8, value: "b-CAP"}, {vt: 8, value: "CaoProv.DENSO.VRC"}, {vt: 8, value: "localhost"}, {vt: 8, value: ""}] }'
4."controller handle" is returned as follows
HRESULT: 0 vntRet: vt: 19 value: "controller handle"
5.Do the follows using "controller handle"
ubuntu@rosservice call /bcap_service '{func_id: 8, vntArgs: [{vt: 3, value: "controller handle"}, {vt: 8, value: "Pro1"}, {vt: 8, value: ""}] }'
6."controller handle" is returned as follows
HRESULT: 0 vntRet: vt: 19 value: "task handle"
6.You can control "pro1" by using  "task handle"

----------------------------------
■Task Start
----------------------------------
ubuntu@rosservice call /bcap_service '{func_id: 88, vntArgs: [{vt: 3, value: "<task handle>"}, {vt: 3, value: "1"}, {vt: 8, value: ""}] }'
func_id:88 // ID_TASK_START
vt:3 // VT_I4 (LONG)
vt:8 // VT_BSTR
----------------------------------
■Task Stop 
----------------------------------
ubuntu@rosservice call /bcap_service '{func_id: 89, vntArgs: [{vt: 3, value: "<task handle>"}, {vt: 3, value: "1"}, {vt: 8, value: ""}] }'
func_id:89 // ID_TASK_STOP
vt:3 // VT_I4 (LONG)
vt:8 // VT_BSTR
----------------------------------
■Task Release
----------------------------------
ubuntu@rosservice call /bcap_service '{func_id: 99, vntArgs: [{vt: 3, value: "<task handle>"}] }'
func_id:99 // ID_TASK_RELEASE
vt:3 // VT_I4 (LONG)

Various operations are available with bcapservice.
Please refer this manual in your ORiN installed folder.
"C:\ORiN2\CAP\b-CAP\CapLib\DENSO\RC8\Doc\b-CAP_Guide_RC8_en.pdf"

zeldamax, 2021-05-31:
Sorry for my late reply. Thank you @DensoWaveRobot. It works now.

closed 2021-05-31 by zeldamax