DENSORobot/denso_robot_ros #36: Failed to clear error (83500372) for COBOTTA

tarukosu, 2020-12-22:
I'm trying to control COBOTTA (Standard OS) with ROS.

When I launch denso_robot_control.launch, "Invalid robot type." error happens.

```
roslaunch denso_robot_control denso_robot_control.launch ip_address:=192.168.0.1
```

After I change robot_name in vs060.launch.xml to CVR038A1-NV6-NNN-NNNMA-NNNA,
"Failed to clear error. (83500372)" error happens.

Log for first command.
```
started roslaunch server http://jolteon:36495/

SUMMARY
========

PARAMETERS
 * /robot_description: <robot name="vs06...
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /vs060/arm_controller/joints: ['joint_1', 'join...
 * /vs060/arm_controller/type: position_controll...
 * /vs060/arm_group: 0
 * /vs060/config_file: /home/ros_catkin_...
 * /vs060/conn_type: tcp
 * /vs060/controller_name:
 * /vs060/controller_type: 8
 * /vs060/invoke_timeout: 180000
 * /vs060/ip_address: 192.168.0.1
 * /vs060/joint_1: 1
 * /vs060/joint_2: 1
 * /vs060/joint_3: 1
 * /vs060/joint_4: 1
 * /vs060/joint_5: 1
 * /vs060/joint_6: 1
 * /vs060/joint_state_controller/publish_rate: 125
 * /vs060/joint_state_controller/type: joint_state_contr...
 * /vs060/port_number: 5007
 * /vs060/recv_format: 292
 * /vs060/retry_count: 5
 * /vs060/robot_joints: 6
 * /vs060/robot_name: VS060A3-AV6-NNN-NNN
 * /vs060/send_format: 288
 * /vs060/timeout: 3000
 * /vs060/wait_time: 0
 * /vs060/watchdog_timer: 400

NODES
  /vs060/
    controller_spawner (controller_manager/spawner)
    denso_robot_control (denso_robot_control/denso_robot_control)
  /
    robot_state_publisher (robot_state_publisher/robot_state_publisher)

ROS_MASTER_URI=http://localhost:11311

process[vs060/denso_robot_control-1]: started with pid [3487]
process[vs060/controller_spawner-2]: started with pid [3488]
[INFO] [1608624696.206270]: Controller Spawner: Waiting for service controller_manager/load_controller
process[robot_state_publisher-3]: started with pid [3493]
[ERROR] [1608624699.855982384]: Invalid robot type.
[vs060/denso_robot_control-1] process has finished cleanly
log file: /root/.ros/log/c1b722e6-4415-11eb-bb67-dc7196a46e56/vs060-denso_robot_control-1*.log
^C[robot_state_publisher-3] killing on exit
[vs060/controller_spawner-2] killing on exit
[WARN] [1608624702.846087]: Controller Spawner couldn't find the expected controller_manager ROS interface.
shutting down processing monitor...
... shutting down processing monitor complete
done
```

Log for second command.

```

PARAMETERS                                                                                             
 * /robot_description: <robot name="vs06...                                                            
 * /rosdistro: kinetic                                                                                 
 * /rosversion: 1.12.14                                                                                
 * /vs060/arm_controller/joints: ['joint_1', 'join...                                                  
 * /vs060/arm_controller/type: position_controll...                                                    
 * /vs060/arm_group: 0                                                                                 
 * /vs060/config_file: /home/ros_catkin_...                                                            
 * /vs060/conn_type: tcp                                                                               
 * /vs060/controller_name:                                                                             
 * /vs060/controller_type: 8                                                                           
 * /vs060/invoke_timeout: 180000                                                                       
 * /vs060/ip_address: 192.168.0.1                                                                      
 * /vs060/joint_1: 1                                                                                   
 * /vs060/joint_2: 1                                                                                   
 * /vs060/joint_3: 1                                                                                   
 * /vs060/joint_4: 1                                                                                   
 * /vs060/joint_5: 1                                                                                   
 * /vs060/joint_6: 1                                                                                   
 * /vs060/joint_state_controller/publish_rate: 125                                                     
 * /vs060/joint_state_controller/type: joint_state_contr...                                            
 * /vs060/port_number: 5007                                                                            
 * /vs060/recv_format: 292                                                                             
 * /vs060/retry_count: 5                                                                               
 * /vs060/robot_joints: 6                                                                              
 * /vs060/robot_name: CVR038A1-NV6-NNN-...                                                             
 * /vs060/send_format: 288                                                                             
 * /vs060/timeout: 3000                                                                                
 * /vs060/wait_time: 0                                                                                 
 * /vs060/watchdog_timer: 400                                                                          
                                                                                                       
NODES                                                                                                  
  /vs060/                                                                                              
    controller_spawner (controller_manager/spawner)                                                    
    denso_robot_control (denso_robot_control/denso_robot_control)                                      
  /                                                                                                    
    robot_state_publisher (robot_state_publisher/robot_state_publisher)                                
                                                                                                       
ROS_MASTER_URI=http://localhost:11311                                                                  
                                                                                                       
process[vs060/denso_robot_control-1]: started with pid [3559]                                          
process[vs060/controller_spawner-2]: started with pid [3563]                                           
[INFO] [1608625036.935672]: Controller Spawner: Waiting for service controller_manager/load_controller 
process[robot_state_publisher-3]: started with pid [3565]                                              
[ERROR] [1608625040.682752542]: Failed to clear error. (83500372)                                      
[vs060/denso_robot_control-1] process has finished cleanly                                             
log file: /root/.ros/log/c1b722e6-4415-11eb-bb67-dc7196a46e56/vs060-denso_robot_control-1*.log         
```

DensoWaveRobot, 2020-12-22:
@tarukosu 
Did you setup step 3 to 5 in this page?
You have to import the URDF of cobotta using ROSConverter.
http://wiki.ros.org/denso_robot_ros/Tutorials/How%20to%20control%20an%20RC8%20with%20MoveIt%21

tarukosu, 2020-12-23:
No, I didn't.
I've copied description files from denso_cobotta_ros package.

I'll try to follow the steps in the tutorial.

tarukosu, 2020-12-23:
I followed the tutorial and set it up, but I got the same error.

DensoWaveRobot, 2020-12-23:
@tarukosu 
When you want to control cobotta(standard OS), please use denso_robot_ros package.
denso_cobotta_ros is a package for cobotta(OSS).

Did you launch as follows?
roslaunch denso_robot_bringup cobotta_bringup.launch sim:=false ip_address:=192.168.0.1 send_format:=0 recv_format:=2

Please connect Remote TP and check that all errors are cleared in cobotta before you launch moveit.

Are there othre errors in cobotta when the error(83500372) occurs? 
You can see the error list by pressing [F6 setting]-[F4 operation and error log].

tarukosu, 2020-12-24:
When I executed roslaunch with Remote TP running, I got the error "起動権をイーサネットにしてください。" (83501028).
After I changed the right to Ethernet, roslaunch was successfully executed and moveit worked.

Thanks for the advice.

closed 2020-12-25 by tarukosu