DENSORobot/denso_robot_ros #41: How to use vacuum gripper using bcap service?

zeldamax, 2021-05-31:
I know I can use the bcap service node to control the electric gripper by doing the following:
Open a new terminal and execute:

`ubuntu@roslaunch denso_robot_bringup <launch file of COBOTTA> sim:=false ip_address:=<COBOTTA's IP Address>`

Open a second terminal and execute:

`ubuntu@roslaunch bcap_service bcap_service.launch ip_address:=<COBOTTA's IP Address>`

Open a third terminal and execute:

`ubuntu@rosservice call /bcap_service '{func_id: 3, vntArgs: [{vt: 8, value: "b-CAP"}, {vt: 8, value: "CaoProv.DENSO.VRC"}, {vt: 8, value: "localhost"}, {vt: 8, value: ""}] }`

Take a look at the returned controller handle value and use it in the following command:

`ubuntu@rosservice call /bcap_service '{func_id: 17, vntArgs: [{vt: 19, value: "<controller handle>"}, {vt: 8, value: "HandMoveA"}, {vt: 8195, value: "30, 100"}] }'`

Now my question is, how can I do the same with the electric vacuum gripper instead of the standard electric gripper? I imagine there is a similar way, in which the bcap service node can control the vacuum gripper. Maybe by using a different func_id?

Also, is there any official documentation on this? I heard there is a pdf file containing information in the ORiN2 installed folder:
"C:\ORiN2\CAP\b-CAP\CapLib\DENSO\RC8\Doc\b-CAP_Guide_RC8_en.pdf"
I took a look at this file, but unfortunately didn't find anything related to the vacuum gripper there.

Thank you very much in advance

zeldamax, 2021-07-09:
Ok so I still haven't solved the problem but I believe that the func_id must stay 17. Because 17 is the function id for "execute command". I changed the command name from "HandMoveA" to "HandMoveVH". Because HandMoveA controls the electric gripper and HandMoveVH controls the vacuum gripper. Also, I changed the parameters from "30, 100" (30 means 30mm opening and 100 means 100% speed) to "100, True" (100 means 100% power and True means vacuum). But it still doesn't work yet. I guess the vt values are still wrong. Maybe vt: 8195 is only for electric gripper parameters? Thank you.

ToshitakaSuzuki, 2021-07-12:
We are sorry for this inconvenience.

>I believe that the func_id must stay 17.

Yes, 17 is correct.

>Maybe vt: 8195 is only for electric gripper parameters?

Yes. As the "HandMoveVH" command takes a direction parameter (VT_BOOL), it is required to change vt: 8195 to vt: 8120 (i.e. VT_VARIANT | VT_ARRAY).
Besides, you need to specify variant type in each element.
So, try to the following command in terminal:
`ubuntu@rosservice call /bcap_service '{func_id: 17, vntArgs: [{vt: 3, value: "<controller handle>"}, {vt: 8, value: "HandMoveVH"}, {vt: 8204, value: "(4, 100), (11, -1)"}] }'`
In this case, vacuum suction pressure is **100[%]** and the motion direction is **Suction** (-1 means True).

For details on the meanings of the parameters, please refer to the "5.2.12.16. CaoController::Execute("HandMoveVH" ) command" in "C:\ORiN2\CAO\ProviderLib\DENSO\RC8\Doc\RC8_ProvGuide_en.pdf".
Although this is an orin provider manual, interfaces is same as b-CAP communications.