DENSORobot/denso_robot_ros #42: Failed to write (83500121) cause by SlvChangeMode(83501032)

suc1, 2021-09-08:
I am using denso SCARA robot arm (RC8 HSR048A1). The ubuntu bionic (18.05.5) is running on Virtual Box 6.1.
Following the tutorial in
[denso_robot_ros Tutorials](http://wiki.ros.org/denso_robot_ros/Tutorials)
It seeems **SlvChangeMode()** cause the problems. The bug always happens.

when run
`roslaunch denso_robot_bringup line8left_bringup.launch sim:=false ip_address:=192.168.0.1`
Get errors:
>[ INFO] [1631121744.315385728]: bcap-slave timeout changed to 16 msec [mode: 0x202]
>[ERROR] [1631121744.564034131]: Failed to write (83500121)
>[ERROR] [1631121744.567636151]: Automatically change to normal mode.
>...
>[ERROR] [1631121745.020626639]:   [2] You cannot execute a command in Slave mode (83501032)
>[ERROR] [1631121745.024187088]:   [1] SlvMove command was executed while the Slave mode was not selected. (83500121)

In TP, the 1st error is:
>You cannot execute a command in slave mode
>[OriginalNumber] 83501032
>[Description] This operation is not available while the robot is in Slave mode
>[Recovery] This command is not available under the Slave mode(range bounded by SlvChangeMode command). Run the command at the outside of the area surround by SlvChangeMode

The 2nd error is:
>SlvMove command was executed while the slave mode was not selected
>[OriginalNumber] 83500121
>[Description] SlvMove command was executed when slave mode was not selected
>[Recovery] Switch to the Slave mode and retry

Because my robot is not the default VS060 in this tutorial, line8left was converted by ROSConverter/AcquireVelAcc

suc1, 2021-09-08:
This bug  is similar to #6 , but I think the root cause is different and it always happen.

DensoWaveRobot, 2021-09-13:
@suc1 
I'm sorry for the late reply.
The error 83501032 is also caused when you change robot to slave mode though the robot is already changed to slave mode. The denso_robot_ros change robot mode to slave mode when it is launching. So I want to know what operation you did before then. Could you please answer the following?
 1. Was the robot normal( not slave mode) when you launched denso_robot_ros? Didn't you change robot mode before?
 2. What operation did you do between when starting on robot to launching denso_robot_ros? If you have sent some command, please let me know.
 3. What is the version of RC8 controller? You can check it on TP. Please open setting and press F2.  
 3. If you have no idea about changing mode, could you share me the wincaps project file that is received from robot just after the error. We can know the detail of the issue from the project file.

suc1, 2021-09-13:
@DensoWaveRobot 
1. The mode is set to Auto. I did not change robot mode manually.
2. I only followed the tutorial and did not send any command to robot.
3. RC8A version 2.5.0
4. Please see attachment for details.
5. I created a new workspace "HSR048A1" to replace "line8left"
[error83501032.zip](https://github.com/DENSORobot/denso_robot_ros/files/7156607/error83501032.zip)

DensoWaveRobot, 2021-09-24:
@suc1 
I sincerely apologize for the late response.
Could you please explain when each errors occur?
We didn't quite understand what operation you did when those errors occur.
We want to know what operation you did after launching HSR and when the error occurs. Especially timing of operation and error is important.

In addition, could you share me correct project file?
The file you send include only ".wpj" file. We need all files in the same folder as wpj to open wincaps project.

suc1, 2021-09-27:
1.  what operation you did after launching HSR and when the error occurs?
I am following the steps in 
http://wiki.ros.org/denso_robot_ros/Tutorials/How%20to%20control%20an%20RC8%20with%20MoveIt%21
  
The bug happened during step "NOT COBOTTA" under "5.Starting MoveIt!" 
`roslaunch denso_robot_bringup hsr048a1.launch sim:=false ip_address:=192.168.0.1`

2. I got a HSR048 definition files(roshsrtest_description.zip) from denso support and still got the same errors (except python errors disappear).
`error83500121.html is roshsrtest's output`
`HSR048A1 is my project fil
[error83501032-2.zip](https://github.com/DENSORobot/denso_robot_ros/files/7236725/error83501032-2.zip)
e`

DensoWaveRobot, 2021-09-28:
@suc1 
I'm sorry for bother you.
The file you sent don't include operation and error log after 9/10. I think the error occurred at 9/13 or later.
In addition, I would like to confirm one thing. About captured Images you sent, "runerror1.png" is correct display if you launched denso_robot_ros. If you execute "roslaunch denso_robot_bringup" at that time, the error occurs. 
Could you let me know when "runerror1.png" and "runerror2.png" have occurred?

suc1, 2021-09-28:
A total files are attached, include PC log file.
[error83501032-3.zip](https://github.com/DENSORobot/denso_robot_ros/files/7245454/error83501032-3.zip)

I canot remember when "runerror1.png" and "runerror2.png" happened. I did not see "runerror1.png" again.

suc1, 2021-10-06:
Could you reproduce this bug under same environments: RC8A, HSR048A1 & [Melodic Morenia (Current Version)](https://rosi-images.datasys.swri.edu/)?

Today when I tried to control UR5, I met network cannot connect problem. I doubt perhaps the network settings in virtual Ubuntu cause this bug.
Tomorrow I will install a native Ubuntu and try again.

karl-exwzd, 2022-01-14:
I'm having the same issue, I can recreate it reliably: I am controlling a standard type Cobotta robot via the `denso_robot_control` package. That works well, however I also need to [actuate the gripper](https://github.com/DENSORobot/denso_robot_ros/issues/21) for which I need to use the `bcap_service` package.

As soon as `bcap_service` starts I get the error that has been originally mentioned in this issue:

```
[ERROR] [1642118543.973920060]: Failed to write (80000900)
[ERROR] [1642118543.999944812]: Automatically change to normal mode.
[ERROR] [1642118544.050668652]:   [2] You cannot execute a command in Slave mode (83501032)
[ERROR] [1642118544.052408843]:   [1] SlvMove command was executed while the Slave mode was not selected. (83500121)
```

Without this `bcap_service` commands cannot be sent to the robot to actuate the gripper.

There is an inconvenient solution: clear the error via RemoteTP. When the error is cleared from the controller the robot can be moved and the gripper actuated via ROS/RViz/MoveIt as expected.

I think the problem comes from the `bcap_service` that attempts to set the robot in slave mode, but at that time the robot is already in slave mode because of `denso_robot_control`. As @DensoWaveRobot mentioned this error is triggered in this situation.

What would be the solution to control the robot and the gripper without this error coming up? One solution would be to programmatically clear the error, if possible, but that feels like a bad patchwork...

ToshitakaSuzuki, 2022-01-17:
A gripper command cannot be accepted while in bcapSlave mode .
In this case,  `You cannot execute a command in Slave mode (83501032)` will happen.

To work both `denso_robot_control` and cobotta gripper, it is required to switch to normal mode before sending a gripper command.

karl-exwzd, 2022-01-18:
@ToshitakaSuzuki Thank you for your reply.

The only way to change modes I've seen is from issues #20 and #29. Basically, I should publish an `Int32` to `/cobotta/ChangeMode`. What value would set the robot to normal mode? I've tried `0`, the error did not appear anymore and could actuate the gripper, but I could not actuate the robot.

Does that mean that every time I want to actuate the gripper I need to switch to normal mode and then back to slave mode for robot control?

ToshitakaSuzuki, 2022-01-18:
> Does that mean that every time I want to actuate the gripper I need to switch to normal mode and then back to slave mode for robot control?
Yes. Unfortunately, this is the limitation of bcap-slave mode.

As you already have mentioned, sending `0` to `/ChangeMode` topic set to normal mode. And sending `0x202` will set to slave mode. The meaning `0` and `0x202` is described in "b-CAP_Guide_RC8_en.pdf" that is included in ORiN2 SDK.