DENSORobot/denso_robot_ros #44: Incorrect parameter when attempting to control a Cobotta robot (80070057)

karl-exwzd, 2022-01-05:
I would like to control a standard Cobotta robot from ROS Noetic. So far I've been successful in integrating it, but now I face an issue without a clear way to fix it.

When connecting to the robot, launching the bringup file, by executing the following

    roslaunch denso_robot_bringup denso_robot_bringup.launch sim:=false ip_address:=192.168.0.1 send_format:=0 recv_format:=2 robot_name:=cobotta

I get the following error:

    Failed to clear error: The parameter is incorrect. (80070057)

There are no error logs on the robot side (checked with RemoteTP) and no further information as to what goes wrong.

I have dug into the source and found that the [`ExecManualResetPreparation`](https://github.com/DENSORobot/denso_robot_ros/blob/d854e94dedab741f19f763186cc0b3ee2dea7f81/denso_robot_core/src/denso_robot_rc8_cobotta.cpp#L41) does not go though. There is also a note for the fuction that calls it: [`ExecClearError`](https://github.com/DENSORobot/denso_robot_ros/blob/d854e94dedab741f19f763186cc0b3ee2dea7f81/denso_robot_core/src/denso_controller_rc8_cobotta.cpp#L77-L82) saying: *Do NOT call on b-CAP Slave*, however the [tutorial on how to control a Cobotta robot](https://wiki.ros.org/denso_robot_ros/Tutorials/How%20to%20control%20an%20RC8%20with%20MoveIt%21) is all about how to enable the *b-Cap Slave* extension (which I did) to control the robot via ROS.

From reading other issues, perhaps it is related to some settings/mode I forgot to change/enable in RemoteTP, but I cannot figure out what it could be.

I do now understand what parameter is incorrect or how to proceed from this point. The communication between ROS and the robot is up, but the error is not helpful... I feel like this is the last challenge before I can move the robot from ROS.

ToshitakaSuzuki, 2022-01-07:
Sorry for the inconvenience.
What is the version of Cobotta?

karl-exwzd, 2022-01-11:
@ToshitakaSuzuki The controller software version is 2.5.1. I had another robot with the controller version 2.8.0 and I could use that one as intended.

Thank you for pointing this out, I will attempt to update the controller software on the first robot.

closed 2022-01-11 by karl-exwzd

bmunkhtulga, 2022-01-13:
Did you guys solve this issue? I am also getting this error when I try to launch MoveIt!. What kind of connection does controller software have here? Is it the one which we can see in the pendant? Mine is 2.7.2. What did you do?

karl-exwzd, 2022-01-13:
@bmunkhtulga First, I would upgrade the robot controller to at least 2.8.0. That would probably solve your issue.

Do not forget to enable the [b-Cap slave extension](https://wiki.ros.org/denso_robot_ros/Tutorials/How%20to%20control%20an%20RC8%20with%20MoveIt%21) on the Cobotta, although you'll be reminded of that if you don't have it in RemoteTP.

bmunkhtulga, 2022-01-13:
Thank you very much for your reply. I solved my issue by upgrading the robot controller from 2.7 to 2.16. I never thought of the controller version being the problem.