DENSORobot/denso_robot_ros #45: How to launch two robots that are same except IP address?

et4141, 2022-01-19:
I have no idea how to launch two hsr065a1 scara robots at the same time(that are same model except IP address).

At the very first time, I launched the robots with different IP address on denso robot bringup package. But when I launched the second robot, the motor of the first robot turned off.

And then, after I made different robot definition file names, such as robot1, robot2, I launched robot1 first. But as soon as I launched robot2, I got a message [Setting the scen for model ‘robot1’ but model ‘robot2’ is loaded] without turnning off the motor of the robot1.

Please let me knoe how to launch two robot at the same time that are same except IP address?

DensoWaveRobot, 2022-01-20:
@et4141 
To control several arm, You have to set different namespace to each node. Could you change robot_name in launch file?

et4141, 2022-01-20:
Which launch file are you talking about? robot name_bringup.launch in denso_robot_bringup package? or denso_robot_control.launch in denso_robot_control package? 
I'm going to control the robots with moveit!.