DENSORobot/denso_robot_ros #46: Hand: Excessive position error

karl-exwzd, 2022-01-20:
While using a Cobotta robot with its standard gripper (controller version `2.8.0`) via ROS Noetic. I tried to grasp an object which obviously stalled the gripper.

The robot triggered the `Hand: Excessive position error` (checked via RemoteTP).

I understand that the gripper will not reach its requested position, but that should not be a reason to trigger an error in this case. How can I prevent this error from occurring? Perhaps there are some hidden settings to allow for large position errors?

ToshitakaSuzuki, 2022-01-20:
Do you use `HandMoveA` command?
This command is for releasing an object or only moving fingers.
When grasping an object, it is recommended to use grasping commands such as:
- `HandMoveAH`
- `HandChuck`
- `HandMoveH`

For details, please refere to [Official User's manual](https://www.fa-manuals.denso-wave.com/en/COBOTTA/007170/).

karl-exwzd, 2022-01-24:
@ToshitakaSuzuki Thank you very much. This solved my issue.

It is not trivial to understand what types of variants need to be sent and what the robot limits are, but I figured them out through trial and error: it seems the gripper effort can only be set between 0 and 20N, I haven't seen specifications related to these limits anywhere...

closed 2022-01-24 by karl-exwzd