DENSORobot/denso_robot_ros #47: [ERROR] [ ]: Invalid argument value (send_format = 0x120)

yo4hi6o, 2022-02-15:
Hello,

Currently, the same error ccurs in my environment and roslaunch does not start.
Do you know what caused this error?

Environment

VS050A3-AV6-NNN-NNN
RC8-VSA-NNNM-NN-NN　　      Ver 1.5.8

ROS            Melodic
GitBranch    Master

1.Create a description file for VS050A3 using the ROS converter.

2.ROS Tool　I tried to get the speed information using Acquire VelAcc, but an error occurred and I could not get it. Therefore, copy the joint limits file of the VS060 file on Github.

3.License registration of b-cap slave to teaching pendant

Operating procedure

Change the teaching pendant to AUTO mode

`roslaunch denso_robot_bringup vs050a3_bringup.launch sim:=false ip_address:=192.168.0.2`

After starting roslaunch, the motor will start.

Execution status

```
yo4hi6o@yo4hi6o:~/denso_ws$ roslaunch denso_robot_bringup vs050a3_bringup.launch sim:=false ip_address:=192.168.0.2
... logging to /home/yo4hi6o/.ros/log/fba5e768-8e14-11ec-993e-54ab3a5cd6b4/roslaunch-yo4hi6o-15450.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://yo4hi6o:44025/

SUMMARY
========

PARAMETERS
 * /joint_state_publisher/publish_frequency: 125.0
 * /joint_state_publisher/source_list: ['/vs050a3/joint_...
 * /joint_state_publisher/use_gui: False
 * /move_group/allow_trajectory_execution: True
 * /move_group/arm/longest_valid_segment_fraction: 0.05
 * /move_group/arm/planner_configs: ['SBLkConfigDefau...
 * /move_group/arm/projection_evaluator: joints(joint_1,jo...
 * /move_group/capabilities: move_group/MoveGr...
 * /move_group/controller_list: [{'default': True...
 * /move_group/controller_manager_ns: controller_manager
 * /move_group/jiggle_fraction: 0.05
 * /move_group/max_range: 5.0
 * /move_group/max_safe_path_cost: 1
 * /move_group/moveit_controller_manager: moveit_simple_con...
 * /move_group/moveit_manage_controllers: True
 * /move_group/octomap_resolution: 0.025
 * /move_group/planner_configs/BKPIECEkConfigDefault/type: geometric::BKPIECE
 * /move_group/planner_configs/ESTkConfigDefault/type: geometric::EST
 * /move_group/planner_configs/KPIECEkConfigDefault/type: geometric::KPIECE
 * /move_group/planner_configs/LBKPIECEkConfigDefault/type: geometric::LBKPIECE
 * /move_group/planner_configs/PRMkConfigDefault/type: geometric::PRM
 * /move_group/planner_configs/PRMstarkConfigDefault/type: geometric::PRMstar
 * /move_group/planner_configs/RRTConnectkConfigDefault/type: geometric::RRTCon...
 * /move_group/planner_configs/RRTkConfigDefault/type: geometric::RRT
 * /move_group/planner_configs/RRTstarkConfigDefault/type: geometric::RRTstar
 * /move_group/planner_configs/SBLkConfigDefault/type: geometric::SBL
 * /move_group/planner_configs/TRRTkConfigDefault/type: geometric::TRRT
 * /move_group/planning_plugin: ompl_interface/OM...
[... 171 lines cut ...]
********************************************************

[ INFO] [1644898149.929385283]: MoveGroup context using planning plugin ompl_interface/OMPLPlanner
[ INFO] [1644898149.929424795]: MoveGroup context initialization complete

You can start planning now!

[ INFO] [1644898151.005178974]: Ready to take commands for planning group arm.
[ INFO] [1644898151.005314032]: Looking around: no
[ INFO] [1644898151.005371093]: Replanning: no
^C[rviz_yo4hi6o_15450_5623600324430190867-7] killing on exit
[move_group-6] killing on exit
[robot_state_publisher-5] killing on exit
[vs050a3/controller_spawner-4] killing on exit
[joint_state_publisher-2] killing on exit
[WARN] [1644898158.184648]: Controller Spawner couldn't find the expected controller_manager ROS interface.
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```

_Originally posted by @yo4hi6o in https://github.com/DENSORobot/denso_robot_ros/issues/29#issuecomment-1039848936_

yo4hi6o, 2022-02-16:
Hello,

Probably an error related to controller version 1 (1.5.8) dependency. There is a description on the second page of the b-cap communication specification, but it seems that the SlvSendFormat and SlvRecvFormat functions were added on 2013-10-08, and the additional functions are reflected by the specifications earlier than this controller. It is thought that the cause is not. If you have any supplements, would you please comment?

ToshitakaSuzuki, 2022-02-16:
That is right. 
Version 1.5.8 was release in 2013-08. Your controller does not support those functions.
Please contact customer service for upgrading a controller.

yo4hi6o, 2022-02-16:
thank you. 
We are already in the process of upgrading the version. I will also report the confirmation status after that here.

yo4hi6o, 2022-08-18:
Since the problem was solved by upgrading the controller, I will close it.

closed 2022-08-18 by yo4hi6o