DENSORobot/denso_robot_ros #52: How to do 24v output for RC8 robot controller from ROS

yo4hi6o, 2022-08-18:
Currently, we are thinking of using a hand that controls the solenoid valve and operates the air chuck.

The controller uses RC8. The model number is VS050A3-AV6-NNN-NNNAN-NNNN. 
Looking at the manual, it seems that it can be output from pin 17 of Hand I/O, 
but could you tell me how to output using ROS?

DensoWaveRobot, 2022-08-26:
Pin17 of hand I/O is + common output. Please use terminal No.1-8. No.1-8 are connected with IO variable[64-71].
https://www.fa-manuals.denso-wave.com/en/usermanuals/005274/

You can operate IO Port or contoroller variable using bcapservice node.

Following sample is for operating P Variable
1.Do the follows
ubuntu@roslaunch denso_robot_bringup sim:=false ip_address:=<ROBOT's IP Address>
2.Do the follows on new terminal
ubuntu@roslaunch bcap_service bcap_service.launch ip_address:=<ROBOT's IP Address>
3.Do the follows on new terminal
ubuntu@rosservice call /bcap_service '{func_id: 3, vntArgs: [{vt: 8, value: "b-CAP"}, {vt: 8, value: "CaoProv.DENSO.VRC"}, {vt: 8, value: "localhost"}, {vt: 8, value: ""}] }'
4."controller handle" is returned as follows
HRESULT: 0 vntRet: vt: 19 value: "controller handle"
5.Getting P[0] variable handle
ubuntu@rosservice call /bcap_service '{func_id: 9, vntArgs: [{vt: 3, value: "<controller handle>"}, {vt: 8, value: "P0"}, {vt: 8, value: ""}] }'
6."Variable handle" is returned as follows
HRESULT: 0 vntRet: vt: 103 value: "<variable handle>"
7.Reading P[0] Value
ubuntu@rosservice call /bcap_service '{func_id: 101, vntArgs: [{vt: 3, value: "<variable handle>"}] }'
8.Writing P[0] Value
ubuntu@rosservice call /bcap_service '{func_id: 102, vntArgs: [{vt: 3, value: "<variable handle>"}, {vt: 8195, value: "100,-44.5,100,180,0,180"}] }'

Various operations are available with bcapservice.
Please refer this documentation in your ORiN installed folder.
"C:\ORiN2\CAP\b-CAP\CapLib\DENSO\RC8\Doc\b-CAP_Guide_RC8_en.pdf"

yo4hi6o, 2022-08-26:
@DensoWaveRobot 

Thank you. Try immediately. I will report the situation here again.

yo4hi6o, 2022-09-08:
@DensoWaveRobot 
As a result of following the procedure, an error occurred stating that it cannot be executed in slave mode.
The timing at which an error is detected occurs during writing.
the procedure is

1
```
rosservice call /bcap_service '{func_id: 3, vntArgs: [{vt: 8, value: "b-CAP"}, {vt: 8, value: "CaoProv.DENSO.VRC"}, {vt: 8, value: "localhost"}, {vt: 8, value: ""}] }'
HRESULT: 0
vntRet: 
  vt: 19
  value: "152"
```
2
```
rosservice call /bcap_service '{func_id: 9, vntArgs: [{vt: 3, value: "152"}, {vt: 8, value: "P0"}, {vt: 8, value: ""}] }'  
HRESULT: 0
vntRet: 
  vt: 19
  value: "153"
```
3
```
rosservice call /bcap_service '{func_id: 101, vntArgs: [{vt: 3, value: "153"}] }'
HRESULT: 0
vntRet: 
  vt: 8196
  value: "244.7381,-37.75574,688.9542,-0.9151886,-14.48771,0.1397858,5"
```
4
```
rosservice call /bcap_service '{func_id: 102, vntArgs: [{vt: 3, value: "153"}, {vt: 8196, value: "100,-44.5,100,180,0,180"}] }'
HRESULT: -2091904974
vntRet: 
  vt: 0
```
After this the following error occurs.
```
[ERROR] [1662612121.404709305]: Failed to write (80000900)
[ERROR] [1662612121.425164998]: Automatically change to normal mode.
[ERROR] [1662612121.495696391]:   [2] スレーブモードでは実行できません。 (83501032)
[ERROR] [1662612121.496659186]:   [1] スレーブモードでない時にSlvMoveコマンドを実行しました。 (83500121)
```
I am operating vs050a3 using slave mode via ROS, but is it not possible to perform write operations in slave mode?
Thank you.

yo4hi6o, 2022-09-08:
An issue was raised [here](https://github.com/DENSORobot/denso_robot_ros/issues/42#issuecomment-1014398851). Change to normal mode and try again. I will also direct the situation here.than kyou.

yo4hi6o, 2022-10-21:
For the time being, I have completed the hand I/O operation by terminal operation, so I will fill it in.

```
// T1
roslaunch bcap_service bcap_service.launch 
```

```
// T2

//  1. Controller Connect & Get Controller Handle

rosservice call /bcap_service '{func_id: 3, vntArgs: [{vt: 8, value: "b-CAP"}, {vt: 8, value: "CaoProv.DENSO.VRC"}, {vt: 8, value: "localhost"}, {vt: 8, value: ""}] }'
HRESULT: 0
vntRet: 
  vt: 19
  value: "controller handle "

// 2. Get Variable Handle

rosservice call /bcap_service '{func_id: 9, vntArgs: [{vt: 3, value: "controller handle ""}, {vt: 8, value: "IO064"}, {vt: 8, value: ""}] }'  
HRESULT: 0
vntRet: 
  vt: 19
  value: "valiable handle"

// 3. Get Valiable

rosservice call /bcap_service '{func_id: 101, vntArgs: [{vt: 3, value: "variable handle"}] }'                     
HRESULT: 0
vntRet: 
  vt: 11
  value: "0"

// 4. Put Valiable Handle

rosservice call /bcap_service '{func_id: 102, vntArgs: [{vt: 3, value: "variable handle"}, {vt: 4, value: "1"}] }'  
HRESULT: 0
vntRet: 
  vt: 0
  value: ''
```

yo4hi6o, 2022-10-24:
@DensoWaveRobot 
Is there a sample code that implements terminal operations like the above with a cpp node?

DENSO-FASupport, 2023-01-26:
Dear @yo4hi6o san,
Thank you for using the DENSO Robot.

I was wondering if I could be of further assistance to you regarding what you were discussing the other day. 
If the issue has been resolved by [here](https://github.com/DENSORobot/denso_robot_ros/issues/42#issuecomment-1014398851), could you close this issue?
I'm sorry to trouble you, but I appreciate your cooperation.

closed 2023-01-26 by yo4hi6o